trustcrypto/OnlyKey-Firmware #26: Multiple Yubikey OTPs

zoidbergthepopularone, 2016-08-14:
As far as I can tell, OnlyKey only supports one Yubikey OTP. Is that so? Since I have multiple Yubikeys, I hoped to replce them all with a single OnlyKey. 

Technically, OnlyKey shouldn't have any trouble with multiple Yubikey OTPs, the only real limitation I can see is that the UI for setting up OnlyKeys just doesn't handle the situation. Or is there a technical reason why only a single Yubikey OTP is supported?

onlykey, 2016-08-15:
Yes this is correct. I did originally test with the capability to simulate multiple Yubikeys but when taking a look at how most people use Yubikeys this did not really make sense. Most people have one Yubikey for everything and possibly a second Yubikey that is a backup. Adding this functionality would be a minor/medium firmware update and same for the chrome app. But we don't want to add features that only one or two people would use so I need to get more information on how many people would really use this.

zoidbergthepopularone, 2016-08-16:
What those users don't understand and you apparently underestimate is this: Given the OTP generation scheme of Yubikeys, if you use the same Yubikey with multiple service providers, you *lose the OTP capability!!!* That is, if providers A and B are set up to use the same Yubikey and you authenticate to provider A, someone can use the captured OTP to present himself to provider B!! I can understand that the users might ignore this problem, but do you want OnlyKey to be known as a device to *enforce* this problem on the users? What would you think of your bank's electronic banking system if it required that you only use a password to log in and that password was limited to four numbers only?

onlykey, 2016-08-16:
So your recommendation to Yubikey users would be to use a separate Yubikey for each application they login to? I will leave this thread open if others are interested in multiple Yubikeys but this is really just one of the advantages that TOTP and U2F have over Yubikey OTP. Given you use a Yubikey for multiple sites and provider A is compromised your Yubikey is essentially compromised for all sites you use your Yubikey for. With TOTP given that provider A is compromised your TOTP key for that provider is compromised but provider B is safe. With U2F given that provider A is compromised your U2F private is not compromised because the provider only has an encrypted key handle.

zoidbergthepopularone, 2016-08-16:
Yes! That is, if they want the security of one-time passwords. (And yes, the server compromise is another reason.)

zoidbergthepopularone, 2016-08-17:
Btw., after reading up on TOTP:

1) TOTP is not a true one-time password. A captured key can be used against a different provider until the time window expires. It is somewhat better than Yunikey OTP where the time window is essentially unlimited, and there is a pretty good protection in that only a part of the token is being transmitted*), but it is not significantly more secure than Yubikey and the disadvantages (e.g. the requirement for synchronization between both sides) are not trivial.

2) The server can generate a valid TOTP in much the same way as is the case with Yubikey. Compromising the server causes exactly the same results in both cases.

I will now study U2F, too, but I am not happy about my findings so far - the flaws you seem to gloss over certainly don't help me to trust in OnlyKey :-(((

*) As long as different providers use a different part of the calculated hash, which is not guaranteed at all. In fact, what I read makes me suspect most providers use the exact same values, which completely destroys any security potentially inherent in the TOTP approach.

onlykey, 2016-08-17:
Response to 1) Something you missed in your research here is that every TOTP account uses a different private that is generated on that website. So your mention that a captured OTP can be used against a different provider is not correct. 

Response to 2) Both TOTP and Yubikey are compromised if the private keys on a provider server is compromised. The difference being that only the TOTP account assigned to that server is compromised, where all accounts for a Yubikey are compromised.

The "flaws" you mention seem to be partially a misunderstanding about how these protocols work. You can find the full details of how these protocols work by reading the RFCs here:
https://tools.ietf.org/html/rfc6238
https://www.ietf.org/rfc/rfc4226.txt

zoidbergthepopularone, 2016-08-17:
So how is it different from Yubikeys? If you use the same private part, a compromise of one server will compromise all. If you use different private parts, a compromise of one server will not mean anything for the others. I should think that proves my point that multiple Ybukey support is important.

onlykey, 2016-08-17:
Yes, this is not a flaw it is how symmetric encryption works there is only one key. Yubikey was designed with this limitation in mind and uses things like YubiHSM so on the server side so that if the server is compromised the keys are not compromised. But it is a limitation of Yubikey OTP and the fact that TOTP uses separate keys is why TOTP has a clear advantage over Yubikey OTP.

onlykey, 2016-10-20:
It has been about two months and there has not been additional interest in this feature. Yubikey OTP with one account is how Yubikeys are intended to be used.

closed 2016-10-20 by onlykey

ExoUNX, 2017-07-14:
I'd also love this feature, but because I like to separate identities. I'd love to use separate yubico OTPs for personal, open-source projects, work, business, and client use. That'd be at least 5 different yubico OTP profiles.

onlykey, 2017-07-14:
So another consideration here is that for most yubico OTP authentication you are actually using Yubicloud. In order to use Yubicloud you have to have a valid YubiKey serial number, you can register a non-yubikey 3rd party device but you still need the serial number. So if you had 5 different YubiKey profiles you would probably need to own 5 actual YubiKeys.

ExoUNX, 2017-07-14:
I myself would be ok with that, it just means I could keep the Yubikeys as backups and the OnlyKey as my primary device.

onlykey, 2017-07-15:
Unfortunately that's not how it works. YubiKey OTP is counter based. So if you have two identical  tokens only one of them has the correct counter value. It's this way intentionally to prevent cloning. So here your backup yubikeys would have the wrong counter and so authentication would fail.

nickalcock, 2017-12-10:
A side note: this is why pam_yubico has (a quite inconvenient) challenge-response mode, where it sends a new HMAC-SHA1 challenge every time you authenticate, then records the old challenge. This is annoying because you have to hit the key twice, but it is still beneficial because if you are not using the yubicloud and your pam_yubico instance cannot reach your authentication infrastructure (perhaps because it's internal to a local network you're not on right now), *you cannot use your OTP any more*.

The real tie here is that each OTP is bound to a single instance of some authentication server somewhere, which shares its secret and knows the last counter values received. Supporting multiple OTPs would not help with this: you'd still need to have multiple auth servers :)

(I think this would be a neat feature -- but I use OTPs constantly and I can't see myself using this.)

ExoUNX, 2017-12-11:
@onlykey Fair enough