trustcrypto/OnlyKey-Firmware #27: Youbikey OTP without a public identity support

zoidbergthepopularone, 2016-08-14:
Yubikey OTPs seem to work fine, except that a public identity is always output. That is not required with the real Yubikey, where the public identity is only optional. OnlyKey doesn't request it when setting up the OTP, but it will still output some public identity when asked to generate an OTP. IMHO, if the Public Identity field is left empty, OnlyKey should take it as a signal that it should ouput only the OTPs without any public identity.

onlykey, 2016-08-15:
Hmm the ones I have used by default include the public ID. Is there an application that requires not having the public ID present?

zoidbergthepopularone, 2016-08-16:
Public ID is optional. If OnlyKey aspires to simulate a Yubikey, it should not enforce requirements that are not enforced by a Yubikey. And it should definitely not be the case that if I leave the public identity field empty, OnlyKey supplies *some* identity on its own (I don't know which , presumably what was stored on the OnlyKey before I reprogrammed it).

Examples of applications not using a public ID include anything which user a regular username with an OTP. I myself implemented this for SlimFTPd server and for UltraVNC, presumably there are other cases.

onlykey, 2016-08-16:
What OnlyKey aspires to do is make 2FA easy. Additional settings that are not required for the majority of applications makes one more thing to confuse people who are not as tech savvy. We will consider this feature but I am still not aware of any use cases other then the custom use case for the software mentioned that you implemented. If you are interested in contributing features to OnlyKey it is open source and pull requests are accepted.

zoidbergthepopularone, 2016-08-16:
I was thinking about that, but unfortunately I would not be able to perform any testing due to the nature of the configuration app :-(

onlykey, 2016-08-16:
@zoidbergthepopularone What is the nature of the configuration app?

zoidbergthepopularone, 2016-08-16:
Chrome Application, something I know exactly nothing about and am not really interested in changing that, I am afraid. My programming interests lie elsewhere.

onlykey, 2016-10-20:
It has been about two months and there has not been additional interest in this feature. Yubikey OTP with the public key included works for most if not all applications.

closed 2016-10-20 by onlykey