trustcrypto/OnlyKey-Firmware #28: Impossible to setup password-only with no terminating character

zoidbergthepopularone, 2016-08-14:
If I setup an OnlyKey slot to:

[x] Label: something
[x] Password: 12345 / 12345
( ) Tab ( ) Return

I expect that only the password ("12345") will be sent by the OnlyKey. Instead, Return is appended to it. That makes it difficult to use two-factor passwords where one part of the password is a fixed password I type and the other part is a random-looking string stored in OnlyKey - this scenario is only feasible if I setup the two parts in order of known-owned, and only if I can always use the same terminating character on all systems (which is not always the case). It should definitely be possible to select "None" as a terminating character.

zoidbergthepopularone, 2016-08-14:
Incidentally, it seems that OnlyKey sends not one, but *two* Returns in this setup, which is even worse as it damages even the use cases where a terminating Return is acceptable (because the other Return will be taken for a real key).

onlykey, 2016-08-15:
For this use case have you considered setting the delay so that you have time to type in what you want after the password? We could change this so that enter is not pressed at the end but as far as I know every use case except the one you mentioned would need to press enter either after password or after OTP entry. And for your use case delay might be sufficient let me know.

zoidbergthepopularone, 2016-08-16:
This issue has multiple aspects with different levels of impact:

1) There is a number use cases where a choice of Tab and Enter is not sufficient. I provided one. Another would be various e-mail login forms which require multiple keypresses between username and password, e.g. https://login.szn.cz, and the same thing can be seen in some desktop application (where you select username, role and password). That makes the current implementation quite limiting. An option to provide a sequence of symbols as terminators would solve all of these cases - the terminator could be empty or it could contain as much as 10 or more symbols, sufficient for all needs.

2) The extra terminator is a serious issue even with a delay, because you have no guarantee of what the user's context will be when that delay expires - the terminator may have serious adverse effects. Note, too, that a delay is not applicable after a password!!! ("Options below only for for 2FA mode").

3) Regardless of any of the above, it is a serious problem that TWO terminating characters are passed after the password. That is definitely an error and even if you decide that #1 and #2 are too much work for too little gain, the two Returns should be removed. 

Note that the UI should reflect what actually happens. If a Return is the default terminator, then it should be selected by default in the configurator. If you leave it empty, as it is now, it leads the user to believe that no terminator will be used, which apparently doesn't happen.

onlykey, 2016-10-20:
It has been about two months and there has not been additional interest in this feature. The desired functionality can be achieved using the delay.

closed 2016-10-20 by onlykey