trustcrypto/OnlyKey-Firmware #30: Firmware Too Large

SteffeyDev, 2016-08-17:
With the new firmware, Teensy (OS X) is saying the file is too large.  I downloaded on both Safari and Chrome, got same SHA 256 sum (5b311d84354152d22c114c9b405a23be4bc95cf7d5ea9445ba5ae5b9265d0e2d).  Sum does not match guide but most likely because creators did not update guide with new sums after uploading new firmware.  See attached picture.

Anyone else having this problem

[image]

onlykey, 2016-08-17:
Yes that is the correct SHA256 sum the quick start guide was updated earlier today. No one else has reported this issue. A user did have a similar issue here but found they were not making good contact with the touch points - https://groups.google.com/forum/#!topic/onlykey/Ed8ylswIQmc

onlykey, 2016-08-17:
The correct file size is 211KB

onlykey, 2016-08-17:
Just double checked teensy loader on Mac and it works fine here 
[image]

SteffeyDev, 2016-08-17:
I didn't realize you had to make the connection before loading the file, that fixed it for me, thanks!

onlykey, 2016-08-23:
Glad this was resolved. Let me know if there is somewhere I might be able to add a note in the quick start guide to make this easier.

closed 2016-08-23 by onlykey