trustcrypto/OnlyKey-Firmware #35: Publish all the files need to compile from source

limpkin, 2016-10-22:
Hello there,

I am not sure why my message was deleted in the official Google Group.
Anyway, reading from your product page: https://crp.to/p/ we can read "open source firmware".
Unfortunately this repository does **not** contain any main(), makefile or the main source code for that matter. All that can be found are **modified open source libraries** without the repository owner own created files.

Please advise.
Regards

closed 2016-10-22 by limpkin

onlykey, 2016-10-23:
Your post was removed from the Google group because the post was obviously not from an OnlyKey user asking for help. It was full of false information and written with malicious intent.

limpkin, 2016-10-23:
My post was made of two things:
- first half asked how to build from source: why should it matter if I'm a onlykey user?
- second half was, exactly like Tim did for onlykey, sharing a tip that may be of interest for onlykey users (using the very same wording that Tim used)...  how is that malicious intent?

onlykey, 2016-10-24:
Your post was made up of two things.

#1 - Make false claims about the product 
#2 - Do so with slanderous intent

You will notice that nothing false or negative was said about your product. 

And then you ask for help compiling after that. I would normally be glad to help but not under these conditions. I am sure if you try harder you can figure this out on your own.