trustcrypto/OnlyKey-Firmware #37: skipping keys

alaska, 2016-11-07:
I've had to lower my OnlyKey typing speed (it's 8 now), but it is still occasionally skipping keys.

With textpad open on my El Capitan Macbook Pro, I reliably get skips at typing speed 10, but with speed 9 they are much less pronounced.

With speed 10, the errors seem to approach 90% in a sample of 10, while with speed 9, I had to run it nearly 20 times before I got an error.  Since I can type almost as quickly on my keyboard and not receive skipped keys, OS latency seems unlikely to be the culprit.  I have a second OnlyKey that I will test with and update this issue.

```OnlyKey Typing Test

Test Username:
12345678901234567890123456
Test Password:
abcdefghijklmnopqrstuvwxyz

Typing Speed 10
---------------
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
123456780123567890123456
abcdefhijklmnopqrstuvwxyz
123567801234567890123456
abcdefghijkmnoprstuvwxyz
123456789013456890123456
acdefghijklmnopqrstuwxyz
1345678901234567890123456
abcdefghijklmnopqrstuvwxyz
1345678901234567890123456
abcdefghiklmnpqrsuvwxz1345678901234567890123456
abcdfghijklmnopqrstuvwxyz
12346789123567890123456
abdefgijklnopqrstuvxyz
123567890123456780123456abcdefghijkmnopqrstuwxyz
12345678901234567890123456
acdefhijklmnopqrstuvwxyz

Typing Speed 9
--------------
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
1234567890123567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
12345678901234567890123456
abcdefghijklmnopqrstuvwxyz
```

onlykey, 2016-11-07:
Hmm I just tried this with the same username/password on El Capitan with speed set to 10 and no loss at all typing this in TextEdit. Although, I have had many issues with El Capitan and have reinstalled about 3 times now to get everything on my system working... The only issue I have seen with OnlyKey on Mac OS is that the pre-boot (File-Vault) screen password entry you have to slow down password entry or it won't work.

alaska, 2016-11-07:
Here are the results from my second OnlyKey

```
1234567901234567890123456
abcdefghijklmnopqrstuvwxyz
1234567890123456789023456
abcdefghijkmnopqrstuvwxyz
1235678901234567890123456
acdefghijklmnopqrstuvwxyz
1234567823456abcdefhijkmnoprstuvwxyz
12345678901234567890123456abcdefghijklmnopqrstuwxyz
123567801234567890123456
acdefghijkmnopqrstuvwxyz
12345678901234567890123456
abcdefhijklmnopqrstuvwxyz12345678901234567890123456
abcdefghijklmnopqrstuvwxyz134568901234567890123456
abcdefghiklmnpqrsuvwxz
1234567890123467890123456
abcdefghijklmnopqrstuvwxyz
```

onlykey, 2016-11-07:
Do you have another computer handy that you can try? If it has the same issue on the other system we may have a bug, if it doesn't this is probably another El capitan bug.

alaska, 2016-11-07:
I'll try and find something.

alaska, 2016-11-07:
Tested in a vim session on an old Asua with an even older version of ubuntu, and it worked.  I'm hesitant to write this off as entirely OSX's problem, but I'm not familiar enough with how the firmware sends keystrokes to debug further.  I *DO* have a rubber ducky laying around somewhere that I can test with, so I'll give that a shot, when I find it.

alaska, 2016-11-27:
Just for an anecdotal reference point, I happened across a yubikey for comparison, and it seems to be able to type more quickly and consistently without errors.  I'll see if I can figure out what speed it types at and tweak the firmware a bit.

onlykey, 2016-11-28:
@alaska I have tried to reproduce this issue but no luck. I am using El Capitan Macbook Pro (Late 2013 model). I have my speed set to 10 and no errors or skipped characters. I am using the latest firmware here: https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v0.2-beta.3

Anyone else experience issues on Mac with type speed set to 10? @rodgolpe @jaroszke

rodgolpe, 2016-11-29:
Unfortunately (...or _fortunately?_) I cannot reproduce this issue either.
OK firmware v0.2-beta.3
OS X El Capitan 10.11.6
MacBook Pro (Retina, 15-inch, Mid 2014)

I tried TextPad and Sublime Text over 100 times each. I also tried varying the delay between username and password output.

closed 2017-03-07 by onlykey