trustcrypto/OnlyKey-Firmware #38: Fails to compile in Arduino IDE

mininggold, 2016-11-24:
Versions:
  Arduino: 1.6.12
  TeensyDuino: 1.31

Error in file: usb_keyboard.c with
    usb_keyboard.c: In function 'deadkey_to_keycode':
    usb_keyboard.c:279: warning: control reaches end of non-void function 
     }
     ^
    expected identifier or '(' before numeric constant

The fix is to above error is to move  
    return 0;      
from line 277 to after the } on 278

Additional errors with 
    keylayouts.h:43:0: warning: "LAYOUT_US_ENGLISH" redefined [enabled by default]

What is the correct procedure for compiling or is there a compile procedure that does not require the Arduino IDE?

mininggold, 2016-11-24:
additional error in usb_keyboard.c:80 with:
    keylayouts.h:237: error: expected identifier or '(' before numeric constant
     #define SHIFT_MASK  0x40

mininggold, 2016-11-25:
Which version of the Arduino IDE and TeensyDuino IDE are required for compiling?

AlbertoFedeli, 2016-11-29:
I also can not make the the code compile. I tested both Arduino v1.6.5 as well as 1.6.12 both with TeensyDuino: 1.31, and both result in several errors.

Obviously we're missing something here. Could we get a quick write up or something about how to compile the code / what libraries we need to modify/add?

closed 2018-02-07 by onlykey