trustcrypto/OnlyKey-Firmware #40: OpenPGP Smartcard Support

activescott, 2017-01-08:
Seems [this is planned](https://github.com/trustcrypto/OnlyKey-Firmware/wiki/OnlyKey-Features#pgp), but I thought I'd create an issue for discussions and people to express support.

Questions: 
* AFAIK the defacto (official?) standard for OpenPGP Smartcard support is g10 Code GmbH's [OpenPGP Card Specification](http://www.g10code.com/p-card.html) (which GnuPG supports). Is the plan for OnlyKey to support implement that standard?
* The OnlyKey-Features wiki page mentions that you're looking for a good PGP C/C++ library. Was the actual GnuPG source and the [GPGME library](https://gnupg.org/related_software/gpgme/index.html) ruled out because it's too big?

Some Relevant References:
* [GnuPG's Wiki on Smartcards](https://wiki.gnupg.org/SmartCard)
* g10 Code GmbH's [OpenPGP Card Specification]( http://www.g10code.com/p-card.html)
* [Yubikey's PGP Docs](https://developers.yubico.com/PGP/) (see "Configure the PGP features of a YubiKey")
* [OpenPGP Card Wikipedia Entry](https://en.wikipedia.org/wiki/OpenPGP_card)

onlykey, 2017-01-08:
@activescott The plan is to support a minimal implementation of OpenPGP to support decryption and signing (private key operations). Public key operations can be handled by client software. Yes, GnuPG is too big and not designed for an embedded device. Currently, ECDSA and ECDH is supported and we are using this for things like SSH and U2F already. The next steps are:

Support for 32 ECC keys and 4 RSA keys (Can be used for GPG or SSH)
Support for RSA using this - https://tls.mbed.org/
Integration with OpenPGP.js

Note: We are not trying to make OnlyKey a smart card so there are not plans to support the OpenPGP Card specification, there are already a lot of products out there that do this. Our approach is to support a driverless version of OpenPGP that communicates via USB HID.

activescott, 2017-01-08:
Thanks for clarifying! So do you expect the user experience to be similar to that of using trezor gpg agent as described [here](https://github.com/romanz/trezor-agent/blob/master/README-GPG.md),  or something else in mind?

onlykey, 2017-03-27:
@activescott We are getting ready to release the first version that supports PGP. The end goal is to be able to support multiple apps for PGP, for now we have this which showcases basic functionality - https://github.com/trustcrypto/python-onlykey

jpathy, 2017-08-21:
@onlykey An interesting approach would be to port gnuk ,which implements openpgp card protocol (https://anonscm.debian.org/cgit/gnuk/gnuk/gnuk.git/tree/README)
The supported hardware(STM32F103) has 128K flash, so i presume the size shouldn't be issue for teensy3.2.
This will seamlessly work with default gpg tools.

onlykey, 2017-08-21:
@jpathy We are currently using the same crypto library as Gnuk - MBEDTLS. The thing about supporting gpg like this is that it requires USB changes that would be doable but not easy. Our approach has been to use USB HID for everything as this doesn't require special drivers like are required for Gnuk and other openpgp card devices.

onlykey, 2017-11-01:
@jpathy  @activescott We just launched a Kickstarter that demonstrates a new approach to OpenPGP support - [https://www.kickstarter.com/projects/1048259057/onlykey-quantum-future-ready-encryption-for-everyo](https://www.kickstarter.com/projects/1048259057/onlykey-quantum-future-ready-encryption-for-everyo)

activescott, 2017-11-20:
👍 I'm in.

M-Pixel, 2017-11-22:
> We just launched a Kickstarter that demonstrates a new approach to OpenPGP support

I don't see the connection. The Kickstarter appears to be about adding a quantum-proof algorithm. It does mention "We are already working on a GPG agent that would integrate with GPG to support multiple apps", but it's under "**possible** integrations we're pursuing", which makes it sound like it's not a promised deliverable, not a funding-tied stretch goal, and therefore not actually a part of the Kickstarter campaign.

I also see the browser extension with plaintext feature which is cute but hardly comparable to SmartCard compatibility which is a lower-level abstraction that would enable use with [several applications](https://gnupg.org/software/swlist.html) and enterprise solutions.

I'm curious @activescott and @jpathy what your particular use-cases are. For me, GnuPG compatibility would be an all-in-one solution for file encryption through my file browser, SSH authentication through numerous applications (in a much more streamlined and broadly compatible way than the Python script), email signing and encryption, and even password manager authentication. It sounds to like it would be, potentially, a lot less effort than trying to enable and streamline all of these integrations individually and directly.

closed 2018-02-07 by onlykey

M-Pixel, 2018-02-08:
I'd like to propose to re-open this issue.

I see that [there's OpenPGP support in Chrome now](https://docs.crp.to/browsercrypt.html), which is a *massive* step forward for versatility that's certainly worth congratulations! It's also a great basis for contributors to implement OpenPGP outside of Chrome.

That said, it's not the same as OpenPGP SmartCard parity. Can I use OnlyKey with Thunderbird + Enigmail yet, or any of the other applications on gnupg.org's list?

I'm not demanding that you make the feature described here, and I don't expect it anytime even this year. I think that keeping the issue open will ensure that there's a single place for gathering feedback, discussion, and votes around this topic until it is completed (or until it's decided in an official capacity that desktop-level OpenPGP compatibility will never be developed).