trustcrypto/OnlyKey-Firmware #59: compiling the firmware

jpathy, 2017-07-10:
Here are some instructions to compile the firmware. Feel free to add it to wiki.

# Steps to compile firmware

* Download Arduino version 1.6.5 from the official website: https://www.arduino.cc/en/Main/OldSoftwareReleases#previous

* Download and install Teensyduino following the instructions from: https://www.pjrc.com/teensy/td_download.html

* Clone the libraries repo inside extraced arduino path:
  ```
  $ cd $ARDUINO_PATH/libraries
  $ git init
  $ git remote add origin https://github.com/trustcrypto/libraries
  $ git pull origin master
  ```
  
* Clone the firmware repo somewhere:
  ```
  $ git clone https://github.com/trustcrypto/OnlyKey-Firmware
  ```
* Few changes to make it work:
  * Backup and replace the following files from OnlyKey-Firmware repo ```$ARDUINO_PATH/hardware/teensy/avr/cores/teensy3/{usb_desc.h, usb_keyboard.c, keylayouts.c, keylayouts.h}```

  * Comment the body of the member function ```set_media```: ```$ARDUINO_PATH/hardware/teensy/avr/cores/teensy3/usb_keyboard.h:90```

* On the arduino UI change USB Type -> "Raw HID" and Board Type -> "Teensy 3.2/3.1".

* Make your own changes, verify and upload.

aveao, 2018-10-23:
Can confirm that this works for beta 6.

Keep in mind that if you comment out / uncomment defines in the main ino file, you'll also need to do the same changes in the onlykey library (in its onlykey.h file) from libraries repo.

bmatusiak, 2020-01-31:
the Teensyduino download for linux requires ardiono src @ v1.8.x,   1.6.5 says unsupported

linux ubuntu 18.04.3 vmware

also ive completed the instructions up until, 

`Few changes to make it work:`

as how or what am i compiling it with,  everything in command line please, i like automated builds for less stress,

but the instructions need more info as to what the next steps are.. 

i was hoping to build this into a DOCKER or some easy script to do it all for you.

nkichukov, 2020-03-09:
It would be great to get official instructions on compiling added into the README/wiki.

bmatusiak, 2020-06-25:
I have this currently working on Ubuntu 20.04

I do __Recommend__ following along with the script by `copy/paste`,  instead of directly running it

https://gist.github.com/bmatusiak/515ae18016b20726da4960591f76baf7/309594572e500e2510e707d8f65efff64da4549a

-----------

this script/process still requires this part
* Comment the body of the member function `set_media`: $ARDUINO_PATH/hardware/teensy/avr/cores/teensy3/usb_keyboard.h:88

the script does open the `usb_keyboard.h` file in `gedit` for your convenience

----------

Pretender1988, 2020-08-15:
Hi together,
I just tried it as descript and got errors. Is the Arduino version 1.6.5  still required? 
with Arduino version 1.6.5  I can not install the Teensyduino libs.

Maybe I just have not seen the actual manual to do that. If you guys could help that would be awesome :)

matcastellan, 2021-11-25:
I was able to compile on Windows using the information above - however, the Teensy loader which appears will not provide the option to load the firmware onto the device, regardless of whether it's in config mode. Trying to load the hex file in the OnlyKey app gives "Error: Firmware block size too large".

From what I can tell searching the internet, it seems like there was a hardware revision for the OnlyKey. Is it still possible to load self-compiled firmware onto the device? And if so, how?

bmatusiak, 2021-12-19:
> I was able to compile on Windows using the information above - however, the Teensy loader which appears will not provide the option to load the firmware onto the device, regardless of whether it's [...]

Flashing of CUSTOM/self-Compiled firmware is DISABLED on 'Production' hardware.
This increases security, for example ,the webcrypt app is restricted and can only be hosted from domain "apps.crp.to"
self-compiled firmware is proven to break features like this. And is why it will never be enabled, This will keep encryption security enforced for all users.

HOWEVER:  I did create a DOCKER environment to build the firmware for DEVELOPER Hardware, found here https://github.com/onlykey/arduino-1.6.5-r5-teensy_127

kalbasit, 2021-12-22:
@bmatusiak where can I get my hands on developer hardware? I want to add Colemak Keyboard Layout support and test it.

bmatusiak, 2021-12-23:
> @bmatusiak where can I get my hands on developer hardware? I want to add Colemak Keyboard Layout support and test it.

@onlykey (tim) may not have any available at this time. because there is a new hardware version being shipped out in a few months

Kulturnilpferd, 2024-01-09:
I'm also bought one and want to code and modify, would be awesome if it would work. Sadly I am stuck with compiler errors maybe a quick guide update would help <3