trustcrypto/OnlyKey-Firmware #62: Illegal use of Mooltipass trademark

mooltipassmachineuser, 2017-08-18:
[image]

We do not allow you to use our trademark, especially to run that kind of adwords campaign which is directly against Google ToS.

onlykey, 2017-08-18:
@mooltipassmachineuser Here are the Google Adwords terms in regards to trademarks - https://support.google.com/adwordspolicy/answer/6118?hl=en

If you contact Google they may be able to help you with this although here is their listed policy 
"An ad can use a trademarked term in its text if either of these conditions is true:
the ad text uses the term descriptively in its ordinary meaning rather than in reference to the trademark
the ad is not in reference to the goods or services corresponding to the trademarked term"

Google Adwords uses dynamic adwords placement where they may replace a term searched for in the ad itself and we do not have control over this.

closed 2017-08-18 by onlykey

mooltipassmachineuser, 2017-08-19:
Please read http://www.azrights.com/media/news-and-media/blog/branding/2014/11/using-competitors-names-in-adwords/

> The issue is whether the ad as a whole suggests a connection with a trademark owner. Would the so-called “reasonably well-informed and reasonably attentive Internet user” be confused into believing that the ad emanates from the trademark proprietor?

> Also, you should ask trademark lawyers who understand the case law surrounding Adwords to consider your proposed ads before you go ahead with them.
(I'm guessing you haven't done so)

By having such an ad, you are tricking users into thinking that onlykey is associated with mooltipass.
We have reported your ad to google.