trustcrypto/OnlyKey-Firmware #63: Enhancement - PAM auth support in Linux

gnujedi, 2017-09-13:
- [x] I searched for this issue before submitting a new issue 
 - [x] I'm not sure if this issue is already possible or not

Example: In order to login to my laptop, it requires my onlykey to be plugged in and unlocked. If removed, or someone else tries to unlock the laptop, it won't allow the attacker access to the system.

jpathy, 2017-10-12:
You could use https://github.com/Yubico/pam-u2f , although it might need some work to be usable with onlykey.

onlykey, 2017-10-12:
@jpathy OnlyKey is a standard U2F device, according to the description this should work with OnlyKey as-is -
"This module implements PAM over U2F, providing an easy way to integrate the YubiKey (**or other U2F compliant authenticators**) into your existing infrastructure."

schlomie, 2018-05-31:
Yubico/pam-u2f works incredibly well with OnlyKey U2F.

I have this rule in /etc/pam.d/pamu2f - `auth    sufficient    pam_u2f.so    cue nullok`

And include it from the common auth pam rule - `auth include pamu2f`

After generating the key at `~/.config/Yubico/u2f_keys` it works from everywhere - login, the lock screen, sudo, etc...

regrettably, 2018-09-24:
Any chance of a write-up on how you accomplished it @schlomie?

schlomie, 2019-06-07:
Install the Yubico pam-u2f libraries and binaries:

- https://github.com/Yubico/pam-u2f
- https://developers.yubico.com/pam-u2f

These are in debian/ubuntu's repositories already (**pamu2fcfg**) arch's (**pam-u2f**)

If your udev rules work, and are able to use the OnlyKey already (as a u2f token e.g. you've set one slot to just be u2f, nothing else,) you should be good to test u2f from the command line:

`$ pamu2fcfg -u <your user>`

After tapping your u2f slot (flashing blue) you should see something like this:

`<your user>:voav4Kt-7dXrsKhAsxWASrxUCGB3D5HEIyNJ89jloHWoZhCjwmtRHr6H9fS-oe1_,04963bfcadf2e5b3b9c6a0058b091d8d78c77fd5d52963291fedc8a1722079eb7aa377d59d92ce8fc8b9116a01ebf320837deb24990cb4e45841aea9b5cb18feec`

If you've made it this far, you are ready to hook into PAM for authentication.

First, pamu2f will look for your key file in `~/.config/Yubico/u2f_keys`

So, create the folder and create the key file:

`$ mkdir -p ~/.config/Yubico && pamu2fcfg -u <your user> >>  ~/.config/Yubico/u2f_keys`

That will, of course write the big long key as seen above to the that text file.  That is where PAM will look to authenticate against your u2f token.

Next, we'll need to configure PAM.

Hope over to `/etc/pam.d`   I have two files, one for required and one for sufficient, so tweak this to your own needs.

**u2f-sufficient**
`auth   sufficient   pam_u2f.so   nullok cue interactive`

**u2f-required**
`auth   required   pam_u2f.so   nullok cue interactive`

For me, I want all system authentication (login, unlock) to require both password and u2f, so in `system-auth`  (I'm on arch.  On debian/ubuntu/whatever, it may vary) I add the line:

`auth   include   u2f-required`

But when using the machine, I have the sudo config only need u2f:
`auth   include   u2f-sufficient`

**ONE NOTE**: PAM changes are effectively immediately.   SO, prove this out with a PAM module that isn't going to lock you out of your machine first.   `sudo` is a good candidate - just make sure you have a su/sudo session open already, in case you missed something and need to tweak.