trustcrypto/OnlyKey-Firmware #64: Passwords are typed when entering challenge pin

M-Pixel, 2017-09-15:
I followed the directions to set up onlykey-agent, generated a key, and copied the public key to the server's authorized_keys list.

When I run the agent with the --connect flag, after scrolling some connection information, a three-number challenge pin is displayed with instructions to type it into the OnlyKey followed by Enter. My assumption is that the instructions are carried over from Trezor where one would be required to press an enter key on the device itself. Regardless, at this point my OnlyKey begins slowly flashing purple - good! However when I press the first number of the challenge pin, if I have store a password on that slot, the OnlyKey types out that password.

I think this is a security flaw, and probably a firmware problem. The firmware should definitely never decide to type out a password when the light is flashing purple.

I have the latest firmware at this time (beta 5).

After pressing just one of the three numbers displayed on the screen, the purple light stops, and after a delay of a few seconds, the connection fails. This might be due to the password entry causing an Enter keypress to be sent from the OnlyKey. I was able to connect succsessfully in the past, before I had increased the typing speed - I believe this is because I had the opportunity to type all three numbers before the password finished typing.

onlykey, 2017-09-18:
@M-Pixel I will see if we can reproduce the issue. The onlykey-agent is still an experimental work in progress. So how this works is that if you press the challenge code, if it is the correct code the signature proceeds, when its done the OnlyKey actually presses enter for you, if any of the three digits entered are incorrect it cancels. So in your first case you say it was flashing purple and when you pressed button it typed out the password, then in the second case the light just stopped? Trying to understand if there was something different in these cases. If the purple light goes off and back to green then the operation has been cancelled so then if you were to press a button it is just back to being a password manager so that would type out the password. 

FYI you probably already know this but if your password was typed in terminal you should change that password.

M-Pixel, 2017-09-28:
After following the new instructions that were just changed in the onlykey-agent readme, I'm able to connect without having a password typed. Thank you for that update.

closed 2017-09-28 by M-Pixel