trustcrypto/OnlyKey-Firmware #65: Enhancement - (configurable) minimum hold time to enter password

M-Pixel, 2017-09-28:
I recall reading on the Kickstarter page that slot \#a is typed by holding the button for 1 second, while slot \#b is typed by holding the button for 5 seconds. This sounds like a great way to prevent accidental password entry in the wrong spot (especially because the buttons are capacitive). Unfortunately, the current behavior is that a momentary press of even less than a half-second causes the slot 1 password to be typed.

A longer minimum would help to make the device more secure. The ability to customize this duration would be nice - potentially even a requirement for those with impaired fine motor skills.

onlykey, 2017-10-12:
@M-Pixel I am on the fence about this one. I think it is more user friendly to do the less that 1 second and more than 1 second approach. We could have an option to enable longer delays for user's that want that. For accidental button press, I think with the OnlyKey case it is harder to accidentally press a button. I have made a change to the firmware that prevents accidentally typing out a password anytime the OnlyKey's LED is fading (SSH, OpenPGP, U2F) and after it stops fading it does not allow a button press for 3 seconds.

M-Pixel, 2017-10-13:
> I have made a change to the firmware that prevents accidentally typing out a password anytime the OnlyKey's LED is fading (SSH, OpenPGP, U2F)

Thank you, I ran into the same problem just now when trying out the OpenPGP script. I see that there is not yet a new release of the firmware. Are there any known issues with the current head?

closed 2019-02-08 by onlykey

rtevans01, 2019-07-25:
I would love to have this feature as well. If I could configure the A slots to have a minimum hold time ot 3 seconds and the b slots 5 seconds, that would be very reassuring. Maybe have the LED light blink red(1), yellow(2), and then green(3) respectively as the seconds pass.

M-Pixel, 2019-07-30:
Thanks for chiming in, @rtevans01.  It's unfortunate that they're closing feature request issues without any official announcement.  It leaves me wondering if they've decided they disagree with the user feedback, or mistakenly think that the issue has been sufficiently addressed.