trustcrypto/OnlyKey-Firmware #66: Enhancement - On-device key generation

M-Pixel, 2017-09-28:
Keys can be generated with OpenSSL and loaded onto the device with the configuration wizard. The ability to instead create keys on the device itself would increase theoretical security of keys, and reduce the prerequisites for using this feature.

I understand that key generation is an intensive process, and the processor on the OnlyKey is nothing like the ones in our phones and computers, so I do wonder about the feasability of this. Would generating X25519 on the Teensyduino take a matter of seconds, minutes, hours, or years?

onlykey, 2017-10-12:
@M-Pixel We can generate ECC keys fairly easily, RSA takes a long time so they are probably not feasible. We can generate ECC keys on OnlyKey in less than a second so this is not an issue. This would be pretty easy to do, but as you say the challenge is to make sure the keys are generated securely. We do have a good RNG so that can be done too. We have a couple other features we want first like stable SSH and OpenPGP in the works. After that I think we could make this happen.

jpathy, 2017-10-12:
Any reason you believe RSA key-generation is not feasible? (Have you tested it?) Smartcards can generate RSA keys of 2048bit in about 9 secs.

onlykey, 2017-10-18:
@jpathy We have not tested it but we use MBEDTLS for RSA, another device called GNUK also uses this and they tried it and said it failed because the embedded device did not have enough memory. Smartcards do RSA in ASICs generally, good for speed but generally can't be verified (not open source). Interesting that we are having this discussion because this is in the news right now where over 1 million smartcard devices may be affected including Yubikey - https://www.engadget.com/2017/10/16/encryption-companies-rely-on-has-serious-flaw/

onlykey, 2018-11-14:
@M-Pixel This is supported in the latest release. Use OKSETPRIV to set an ECC key where all values are 0xFF and a key of the specified type will be generated in the specified slot.

closed 2018-11-14 by onlykey