trustcrypto/OnlyKey-Firmware #68: Keyboard layout for "Mac(BookPro) French"

ebouchut, 2017-12-26:
Hi,

I'm testing the **color OnlyKey (OK) standard edition** (with the stock **firmware v0.2-beta.5**) on my **MacBookPro with a French keyboard layout**.

I like my OK with all the features it provides. It really is a breakthrough for daily use that no other key AFAIK provides to date. 

I configured it to use the **French** keyboard layout  which is slightly different from the one of the MacBookPro with a French keyboard that I'm using. 

This has 2 consequences. A mismatch both during input and output. 

First, it makes filling in the slots a really cumbersome and error prone operation as I need to memorize and use a different mapping than the one I have on my keyboard in the front of my eyes.

Secondly, each time OK populates a form field automatically, it does not write the expected character due to the keyboard layout mismatch.
To give you an example, I recently made a backup of the OK following the user guide and discovered it cannot be restored.
Why?
When OK types the header and footer in the backup data field, it uses a '§' instead of the required minus sign '-'  `§§§§§BEGIN ONLYKEY BACKUP§§§§§`.
This also does occur everywhere else and for a bunch of other characters  (in the "body" for instance) in the backup data field. I discovered that this makes the backup ... useless unless processing it with a script to apply the correct mapping!

This is the reason of me asking for support here.  
Would you please provide us with an upgraded firmware offering the ability to use the **French Mac(Book(Pro))** keyboard layout.

Thank you for your time and for the OK.

onlykey, 2017-12-27:
@ebouchut Do you know where I could find details on what keys are different on AZERTY keyboard and the Macbook Pro AZERTY keyboard?

ebouchut, 2017-12-28:
Hey @onlykey,
Thanks for taking this issue into consideration. 
I will do my best to find you good links about this.

ebouchut, 2017-12-28:
For now, I did not find any comparison with the differences between the 2 keyboard layouts.
However using _Apple keyboard viewer_ here is what I can get quickly.
I took a few snapshots of the layout when pressing the most common keys.

## No Key Pressed

[image]

## Shift Key Pressed

Pressing the `Shift`key gives access to:
* Hash sign: `#`
* Undescore: `_`
* Star: `*`
* Slash: `/`
* Plus sign: `+`
* Percent sign: `%`
* Dot: `.`
* Question mark: `?`
* Greater then sign: `>`

[image]

## Option Key Pressed

The `Option` key is also known as `Alt` key.
There you will find the:
* curly quotes `{` and `}`
* tilde `~` FYI Two keys need to be pressed at the same time to get a tilde : `~` and `Space`

[image]

## Control Key Pressed

Nothing really useful in my opinion here.

[image]

## Option + Shift Keys Pressed

When pressing the left `Option` key and the left `Shift` keys at the same time.
There you will find the:

* backslash: `\`
* square brackets `[`, `]`

[image]

Some keys have an orange background, [here is why](https://apple.stackexchange.com/questions/179699/what-does-the-orange-keys-mean-in-keyboard-viewer).

ebouchut, 2017-12-29:
Should you need any key code, just let me know so I can find them using the program [Key Code](https://itunes.apple.com/fr/app/key-codes/id414568915?mt=12) , we may use a Google Doc for this purpose.

ebouchut, 2018-01-06:
The table below displays the actual character OK outputs when using a key on the MacBookPro using the AZERTY French keyboard layout available in firmware v0.2-beta.5.

Say for instance I edit a slot and type `@` in the username field,
then when OK outputs the username a `ø` is displayed instead.
To make OK output an at sign (`@`) I need to type `<` in the field.

 Input (what I type in a field)  | Output (what OK really displays) 
 -------- | --------
 <         |  @      
 \>        |  #      
 @        |  ø      
 \-        |  §      
 °        |  !      
 \#        |  "      
 \=        |  \-      
 \+        |  _      
 !        | =      
 §        |        
 é        |        
 è        |        
 ç        |        
 à        |        
 ^        | Ç      
 ù        |        
 `        | «<Space> 
 ¨        |        
 \*        | \`      
 %        | %      
 £        |        

I did not managed to find out out to output a star `*`.

onlykey, 2019-04-24:
We have been a bit busy with other features but we do want to support additional keyboard layouts. Would you be able to do a pull request for this? Here is an example of a pull request to add a keyboard layout - https://github.com/trustcrypto/OnlyKey-Firmware/pull/90