trustcrypto/OnlyKey-Firmware #70: Cannot type å ä or ö using the swedish layout.

jacwib, 2018-01-16:
Using the swedish firmware, i cannot type the swedish characters å ä and ö.

This is a problem because one of my passwords use those characters.

Example:

Setting password to "abc123åäöÅÄÖ" instead prints "abc123".

onlykey, 2018-01-16:
@jacwib When you say swedish firmware do you mean you have set in your preferences to use the swedish keyboard layout?

jacwib, 2018-01-16:
@onlykey I have it set both in onlykey settings and my system settings.
If its related, the setting in onlykey still says united states (default) after setting it to swedish.
(But my symbols work properly, and they wouldent if it was set to us, so its still using the swedish layout, it just cant type å ä or ö)

ToasterKTN, 2018-01-27:
Same problem with GERMAN_SWISS / GERMAN_MAC
@äöü and maybe other special Chars are not correct.
Getting “ instead of @ for example

douglaswinter, 2018-06-13:
The hash character (#) in UK layout simply does not print.

onlykey, 2020-11-20:
Unicode characters are not supported at this time

closed 2020-11-20 by onlykey