trustcrypto/OnlyKey-Firmware #72: Config mode

subseaalex, 2018-03-02:
Cannot enter into config mode on v0.2-beta.5 running Arch.
Please advise how to update firmware.

Rgds,

onlykey, 2018-03-02:
Did you hold button 6 down for 5+ seconds.?

https://docs.crp.to/usersguide.html#loading-onlykey-firmware

subseaalex, 2018-03-03:
Yes, I did.
I can't get the light to switch off after 5+ seconds. Everything works fine in chrome app and in onlykey-cli except I can't get in into config mode.

deadi, 2018-03-08:
-No config mode for me
-LED function is arbitrary on Macbook Pro. 
-LED working on old Macbook, iOS and Linux
-after holding button 6 for 5+ sec an rentering PIN, configuring PGP key to slot1, saving as Subkey1, backup key, decryption key: ERROR NOT IN CONFIG MODE, HOLD...
- App v5.0-beta.0
- Firmware v.02-beta.6c

[image]

onlykey, 2018-03-08:
@deadi Whenever you hold button 6 5+ seconds the LED turns off, then you enter your PIN again and then device flashes red indicating its in config mode. Keep in mind if you purchased the international travel edition there is no config mode, this feature is only in the standard edition.

deadi, 2018-03-08:
@onlykey  I installed FW OnlyKey_Beta6_SE_Color.cpp.hex on TravelEdition OnlyKey

[image]

I can really not tell and troubleshoot efficiently, but sometimes LED are working, but most times they aren't.

deadi, 2018-03-08:
@onlykey and if they work, i cannot unlock OnlyKey. Button not responding

[image]

onlykey, 2018-03-08:
@deadi I am not clear what you are having an issue with here, configuring a PIN, entering your PIN after initial configuration, or entering config mode after entering your PIN.

deadi, 2018-03-08:
@onlykey 
Thanks so much for your quick responses. 

1. firstly, the LED won't light up at all times I insert it into USB-slot, sometimes but rarely (I have no idea why) they light up. As far, it seems it never lights up when OSX OnlyKey App is running.
 
2.  I upgraded FW OnlyKey_Beta6_SE_Color.cpp.hex on TravelEdition OnlyKey with TeensyLoader, because i want to store keybase PGP private key on it, hence i need config mode. I was hoping a FW Upgrade will also solve the LED problem.

3. without fully functioning LED I hardly can tell if I managed to enter config mode

4. if (somehow) the LED rarely work, i managed to unlock OnlyKey partly, holding button #6 for 5+ sec, the LED won't go off, neither after re-entering PIN there is flashing

- I was able to configure a PIN 
- OSX App is v5.0-beta.0
- I upgraded Firmware v.02-beta.6c

deadi, 2018-03-14:
@onlykey 
I managed to do all the mentioned configs on Ubuntu 16.04.4 LTS. with OnlyKey DesktopApp. Also the LED is working at all times.
However on OS X 10.11.6 i have stopped trying, it's useless and chainsawing my nerves. 
Thanks anyhow for your support. It's a great tool.