trustcrypto/OnlyKey-Firmware #75: U2F Feature does not follow standards

cjoster, 2018-06-07:
U2F Feature responds affirmatively to keyhandles it did not create. A check-only authentication succeeds, and a user-presence-test simply hangs.

According to this[0] document:

```0x07 ("check-only"): if the control byte is set to 0x07 by the FIDO Client, the U2F token is supposed to simply check whether the provided key handle was originally created by this token, and whether it was created for the provided application parameter. If so, the U2F token must respond with an authentication response
message:error:test-of-user-presence-required (note that despite the name this signals a success condition). If the key handle was not created by this U2F token, or if it was created for a different application parameter, the token must respond with an authentication response message:error:bad-key-handle.```

A basic challenge can be obtained here[1]:

```{"challenge": "Pa3eucFQrH-5c9CAEdGESJiIW9po_Sozs6EfPeYN3nM", "version": "U2F_V2", "keyHandle": "Eu-I54B3MeKSdrOXKzt5e0L0MGtoKUoCFg77ZK7EefIVL0l-Tac2uJvV20kk1x9Ems4EYgtYgiVfNK3nFXhUGA", "appId": "https://demo.yubico.com/app-identity"}```

Command to perform check-only authenticate:

```u2f-host -aauthenticate -o https://demo.yubico.com -t < challenge```

Command to perform full authenticate: 

```u2f-host -aauthenticate -o https://demo.yubico.com < challenge```

[0] - https://fidoalliance.org/specs/u2f-specs-1.0-bt-nfc-id-amendment/fido-u2f-raw-message-formats.html
[1] - https://developers.yubico.com/libu2f-host/

onlykey, 2018-06-08:
Yes, you are correct. This was intentional as checking prior the validity of a keyhandle prior to a user providing physical presence is a security concern. More info here https://bugzilla.mozilla.org/show_bug.cgi?id=1440044

The first security concern was that if any web site can check if a keyhandle is valid without a user's physical presence this could permit an unattended U2F key to be repeatedly checked to conduct a power analysis attack. A second concern was that if any web site can check if a keyhandle is valid without a user's physical presence this could be used to identify and track a user.

We have a solution that will address both concerns and be included in the next firmware release. Using two separate keys, double encryption on the keyhandle the outer layer of encryption used for checking. The outer layer key derived using application ID so that only the correct site can check.

closed 2018-06-08 by onlykey

cjoster, 2018-06-08:
Security concern or not, how do you expect developers of compatible tools do write software that supports your device if the standards are open to whims. I agree with you that it's an interesting problem and a concern, but as it sits, the OnlyKey is wholely incompatible with pam-u2f, and only works if and only if the onlykey is the only configured token (or at least the first), and is the one you want to login with. Out of the box, unconfigured even, or configured but locked makes a machine configured for u2f login unusable. Perhaps a compatibility mode?

onlykey, 2018-06-09:
@cjoster Not sure but it seems like you missed this part of what I said "We have a solution that will address both concerns and be included in the next firmware release"

cjoster, 2018-06-10:
> Not sure but it seems like you missed this part of what I said "We have a solution that will address both concerns and be included in the next firmware release"

Yeah, I caught that part. And calm down a little on the hostility--I also caught the, "this is intentional" part too, so you can't fault me too much for being a little annoyed at paying for an intentionally broken product.

Can I beta test this feature? Is there relevant discussion going on somewhere I can participate in? Can I contribute somehow? Do we have a release date for this next firmware release? Do you want to chime in over [here with some suggestions?](https://github.com/Yubico/libu2f-host/pull/101)

I want to help with this project as I think at least part of it is headed in the right direction.

onlykey, 2018-06-11:
@cjoster I apologize if you felt that was hostility, it just sounded like from your response that you may have overlooked that part as you mentioned a compatibility mode. The OnlyKey still works with multiple U2F devices currently, it just has to be the first U2F device there is a note of this in the user's guide. It would be great if you want to beta test here is the current debug firmware https://github.com/trustcrypto/OnlyKey-Firmware/tree/master/OnlyKey_Beta/Debug%20Firmware firmware with debugging enabled should only be used for testing.