trustcrypto/OnlyKey-Firmware #76: Discussion: Suggested Enterprise Use for OnlyKey

RainmanJS, 2018-07-03:
I recently purchased an OnlyKey for personal use, I can see quite a few use cases for it within an organization.  I wonder though, what the CryptoTrust Team's vision would be?  Perhaps this isn't the correct place to ask the question, if so, please let me know.

For instance, in issue #40, it seems that use as a smart card isn't a current priority.  In an organization with varying levels of technical expertise, how would you recommend using the OnlyKey?  Let's say a typical organization with mostly Windows Desktops in Active Directory, some iOS, Android, and MacOS systems, as well as some externally supported SAAS systems as well as local database servers, some windows, some Linux.

If I was to try to sell OnlyKey as a solution over Yubikey, what would be my argument?  Or is that not how you see the product?

timetoseek, 2018-09-07:
Many Corp users already are using apps like LastPass and Keeper, so OnlyKey would be a great complementary tool...

I love OnlyKey, have been using it a lot.  However, my only real concern is how simple it will be for the average Corporate user?  

Please keep in mind that many (most?) Corporate users don't even have admin rights over their on PC's, which means the IT department would need to do the initial install of the app (as well as any firmware updates) prior to deployment to users.  

A simple training video would be helpful for the users on how to set up the slots, etc.

RainmanJS, 2018-09-07:
I agree on many of your points, the last update CryptoTrust sent indicated that they were building a provisioning method into the next release of the firmware:

**- Enable enterprise provisioning**

Setting certain values on first use is now permitted to enable enterprise provisioning of OnlyKeys. For example, a backup file or passphrase can be set on first use after setting a PIN. This makes it possible for our enterprise business customers to easily provision devices as follows:

- Enterprise provisions a file (securely encrypted with a unique passphrase) for each user that contains the settings for a user's OnlyKey
- User sets their own PIN on a new OnlyKey
- User loads encrypted file and passphrase provided by enterprise
- Device is now configured with required enterprise settings/accounts and ready to use

timetoseek, 2018-09-07:
Thank you, this strategy sounds good! Additionally, an update to the Chrome app would be helpful as well. In some areas the app is intuitive, while in other areas it needs a bit of polish. For example, please see my post on issue [](url)https://github.com/trustcrypto/OnlyKey-Firmware/issues/78

onlykey, 2018-11-14:
@RainmanJS @timetoseek Thanks for starting this discussion. Yes, we definitely do need a good training video. Things have changed so much in the latest release that making a video would have meant re-making the video with this release.   For enterprise use our latest features would enable enterprise provisioning (i.e. Application used by Administrators to generate OnlyKey backup file containing user's accounts and corporate settings). The user could just set pin code, set backup passphrase (given by Admin), and load the OnlyKey backup file to set up all accounts. Additionally, the OnlyKey desktop app could query a server on a set schedule to upload backup file (in case a user loses key there is a current backup) and the Admin would have the ability to push down backup files to add, update, remove, or restore accounts. Remote zeroization would also be possible in this way.

So that's a basic outline of what could be done, but the apps to do this don't currently exist. For your question "If I was to try to sell OnlyKey as a solution over Yubikey, what would be my argument?" here are some things:
- Open source vs. closed source
- Physical security (Yubikey can be used if lost or stolen)
- Multiple accounts 24 slots vs. 2 slots
- Manage passwords not just 2FA
- Upgradable firmware, we continue to release new firmware with new features.

timetoseek, 2018-11-27:
To be quite honest, your latest firmware release is a game changer!  As a matter of fact, I have altogether stopped using YubiKey and now exclusively using OnlyKey.  I am still doing some testing, but do plan to recommend OnlyKey to our IT professionals for deployment. I do believe NFC would still be a very good addition to OnlyKey at some point, but I see that as an enhancement rather than a critical need.

Interesting side note: As I no longer use YubiKey, I tried to destroy it before disposing of it, but could not break it, could not bend it, could not cut it!  I guess that is one great feature of YubiKey, basically they're indestructible!
[image]

onlykey, 2018-11-27:
@timetoseek LOL this is great. I am really glad it is working out well for you and thanks for recommending it. One fun thing to try with your Yubikey would be a hardware tear down if you are going to toss it anyway. These guys did a teardown of the Yubikey 5 - http://www.hexview.com/~scl/neo5/
But I don't think anyone has done one of the YubiKey U2F. For the old version just soaking it in acetone would remove the plastic, now I think there have been some improvements made so you would need a different chemical solvent.

timetoseek, 2018-11-28:
>These guys did a teardown of the Yubikey 5 - http://www.hexview.com/~scl/neo5/

Excellent, thank you for the link!

> But I don't think anyone has done one of the YubiKey U2F. For the old version just soaking it in acetone would remove the plastic, now I think there have been some improvements made so you would need a different chemical solvent.

I will give it a try, I have some solvents left over from a home remodeling project, so will gather those chemicals and post pics of the torn down YubiKey (or, my chemically burned fingers, if things don't go well)

timetoseek, 2018-11-29:
**Ability to save file directly to OnlyKey:** 
Ability for the user to store a small data file on the OnlyKey device. The reason I mention this is because for my webmail access, I not only need to provide a username and password, but also insert a usb device where I saved a small text datafile (~1K).  I do not know if anyone else has this type of authentication system, but if they do then they may also benefit from the ability to save the small text file to OnlyKey (or, they can save their private PGP key, etc).  I don't know if it's possible to do this since OnlyKey is basically recognized as a keyboard, and not a drive, but I thought to make this suggestion just in case...

Kraynyan, 2018-11-30:
> **Ability to save file directly to OnlyKey:**
> Ability for the user to store a small data file on the OnlyKey device. The reason I mention this is because for my webmail access, I not only need to [...]

Even not with the case you provided, I'd love a feature like this, even if it takes up a full, nemerical slot.  E.g 1a & 1b

onlykey, 2018-11-30:
@timetoseek It would be possible to do this, but as you mentioned OnlyKey is not a drive so this would require a new feature in the app to load the file and a new feature in the firmware to store the file. If we did this we would probably have the information typed back out by pressing a button for 5+ seconds similar to the backup.

timetoseek, 2018-11-30:
>If we did this we would probably have the information typed back out by pressing a button for 5+ seconds similar to the backup.

Would it be possible to have OnlyKey function as both a usb keyboard as it now is, and also act like a thumb drive, with small data storage (say 1MB max)?  The reason I mention it is because I don't think you can type out the key.dat file, even though it is a text file, because user needs to browse to it on the thumb drive and loaded to the webmail server for validation (i.e., there's no place to copy/paste the key.dat file, must be loaded to the mail server every time I log in)

If you do end up typing it out, that would be OK, the user can just paste it into a text file and save it, and also would be true for one's private PGP key, etc...

onlykey, 2018-11-30:
@timetoseek No not really, the device has a total of 256K.

timetoseek, 2018-12-03:
Quick note, as we are discussing features for enterprise users, it may be best to disable the "_auto launch on system login_" feature of the OnlyKey app (except for Admins, who would frequently use the app).  There is a thread about this item in the OnlyKey app section, so I'll just add a link to it here:
https://github.com/trustcrypto/OnlyKey-App/issues/89#issuecomment-443277020

onlykey, 2018-12-03:
@timetoseek Auto launch is required if a user wants to use TOTP. Without the app running TOTP will not work so that is why that feature is there.

freddyyeddy, 2018-12-03:
Maybe add the option to run a background process that sets totp time?

onlykey, 2018-12-04:
@freddyyeddy That's eventually the plan, its got to work on Windows / Mac / Linux though.

timetoseek, 2018-12-24:
Please note that Windows 10 **Enterprise** Edition is starting to be deployed in all new laptops of many organizations (including ours, for several thousand users already, and many more planned for the coming months).  In all new deployments, BitLocker is enabled by default (including the pre-boot PIN option).  

FYI, there is a thread https://github.com/trustcrypto/OnlyKey-Firmware/issues/81#issuecomment-449608798 related to my testing of Win 10 Enterprise over the past week. I have run in to a few issues, which are not critical but nevertheless can be impediments to enterprise-wide adoptions of OnlyKey...

timetoseek, 2019-05-14:
This topic has some overlap with the "Too hard to find" thread, so adding a link to it:
https://github.com/trustcrypto/OnlyKey-App/issues/85

M-Pixel, 2019-05-15:
I wrote a pair of programs for my organization that provision and use an OnlyKey to decrypt credentials that are downloaded from a server.  We mail a client the OnlyKey and communicate the passcode to them over Signal, then they plug it in and run our program to gain access to a VCS.

onlykey, 2019-05-17:
@M-Pixel This sounds like a good use case. There are some companies mailing OnlyKey with a head-less server, for example a pentesting device. Then securely communicate the PIN to the client and they use the OnlyKey to enter a strong FDE password. I have been thinking of documenting this in our docs, anything you can share in regards to your application?

closed 2021-11-26 by onlykey