trustcrypto/OnlyKey-Firmware #77: Keyboard Layout GERMAN_SWISS

dunmark, 2018-08-18:
Hi

When using a MacBook with a QWERTZ Swiss-German layout, there are issues with some characters being mistaken such as the **@**. OnlyKey Layout is set to GERMAN_SWISS.
This is what the layout looks like:

[image]

when pressing **alt**:

[image]

when pressing **shift**:

[image]

Can you please take a look into it?

Thanks

onlykeyuser, 2018-11-19:
Please fix the keylayouts.c in the source code 

at line 884 (LAYOUT_GERMAN)
from
ASCII_40=KEY_Q + ALTGR_MASK;// 64 @
to
ASCII_40=KEY_L + ALTGR_MASK;// 64 @

at line 1103 (LAYOUT_GERMAN_MAC)
from
ASCII_40=KEY_Q + ALTGR_MASK;// 64 @
to
ASCII_40=KEY_L + ALTGR_MASK;// 64 @

German Mac keyboards take ALT- lowercase L to type a "@".
Thank you!

onlykey, 2019-04-24:
We have been a bit busy with other features but we do want to support additional keyboard layouts and make sure the existing ones are correctly mapped. Would you be able to do a pull request for this? Here is an example of a pull request to add a keyboard layout - https://github.com/trustcrypto/OnlyKey-Firmware/pull/90

tafli, 2020-04-20:
@dunmark I'm not sure, whats best for Swiss_German Layout. On Mac, the German_Swiss Layout is different from other OS. Like curly braces on a Mac are on ALT+8/9. On normal Swiss German keyboards however, they are on ALT+ä/$.

Probably it must be distinguished between OSs as well? Or add a German_Swiss_OSx?