trustcrypto/OnlyKey-Firmware #79: Keyboard Layout HUNGARIAN

lcsaszar, 2018-10-31:
Dear developers,

Could you add Hungarian keyboard support? How could I enter special characters until this is done? When I enter characters on a Hungarian layout keyboard, they appear correctly on the configuration page, but the device returns different characters (obviously those of the US keyboard layout).

Thanks

takd, 2019-03-29:
Yes, please! I tried every possible available keyboard layout, but all of them lack the ability for basic non-alphanumeric characters to be typed on a Hungarian keyboard. I would also be glad to work on this myself, if i get sufficient instructions from the developers.

onlykey, 2019-04-24:
https://github.com/trustcrypto/OnlyKey-Firmware/pull/90

closed 2019-04-24 by onlykey