trustcrypto/OnlyKey-Firmware #81: BitLocker issues with new firmware

timetoseek, 2018-11-26:
After the firmware upgrade, there is some unusual behavior on PC's with BitLocker encryption, as follows:

**_Testing on Windows 7 Enterprise (Service Pack 1)  PC (HP EliteBook)_**
If OnlyKey is inserted _prior_ to turning on the laptop, then I am able to input pin and use OnlyKey normally.  However, if OnlyKey is inserted _after_ turning on the laptop, then OnlyKey does not work. 
I am still able to type in my pin on the device, and it glows green, but nothing is passed to the PC when I press any number on the device.
Also, after boot and successful login into Windows--  if I remove OnlyKey, then reinsert the device, an error message will show up (please see screenprint, below)
[image]
Maybe after the firmware upgrade, I need to log in as Admin to use OnlyKey on Win7 PC?  I can't test this since I don't have admin rights on the Win7 PC.

**_On a Samsung Ultrabook PC running Windows 10 Pro (v1809)_**
If the Only Key is inserted _prior_ to power up, then it is recognized by Windows and I am able to press the key on the device which successfully passes thru my password to unlock BitLocker. However, if I insert the OnlyKey _after_ power-up, then nothing is passed thru to the PC.  I am still able to type in my pin on OnlyKey, but nothing is passed thru to the PC 
(if I insert OnlyKey after power up, then I must wait to login to Windows for OnlyKey to work properly. Logged in as Admin)

**_Linux Mint 19, running on a MacBook_** (yes, Linux on a MacBook, no choice, as it's no longer supported by Apple to run MacOS)--
My entire drive is encrypted and requires a key at startup.  I've tested this at some length and in summary, OnlyKey worked flawlessly in Linux, from boot, all the way through various testing.

_One other note, as others have observed, the Chrome browser app is no longer compatible..._

onlykey, 2018-11-26:
@timetoseek There was an early issue with some Win7 systems. This has been addressed and the firmware file updated here - https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v0.2-beta.7

More detail here - https://groups.google.com/forum/#!searchin/onlykey/windows$207%7Csort:date/onlykey/xgjGn-Eb_hk/0tyuK__dAQAJ

For the Chrome app you mention " as others have observed" can you point me to where others are having issues? The Chrome app was updated on the Chrome web store right after the firmware release last week. Chrome apps should update automatically but if you are having issues you can add it here https://chrome.google.com/webstore/detail/onlykey-configuration/adafilbceehejjehoccladhbkgbjmica

timetoseek, 2018-11-27:
Thank you for the update, and really for the great work you guys are doing with OnlyKey, it has now replaced all the other devices/U2F/etc that I've been trying out for the past year.  Great job!  
With that said, a few notes:
>There was an early issue with some Win7 systems. This has been addressed and the firmware file updated here - https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v0.2-beta.7
> 

I did install the latest firmware (as far as I can tell). Here is the screen print:
[image]

If there is a newer version of the firmware? (I searched but was not able to locate any newer firmware). Or, are admin rights required?  Unfortunately I do not have admin rights to the Win 7 laptop and not able to test.

> For the Chrome app you mention " as others have observed" can you point me to where others are having issues? The Chrome app was updated on the Chrome web store right after the firmware release last [...]

The Chrome app did not update automatically in the Win 7 laptop I used for testing.  I thought I read an issue about this on the Google group that others are experiencing this, but I may have been mistaken..  In any case, I deleted the app at this point from the Win 7 laptop and not able to do further testing.  I am only using the Desktop app on my Win 10 laptop, which works "OK" but there are a few issues with it, as follows:

Must remove/insert OnlyKey device, otherwise non-responsive, as shown in the following screen print:
[image]

Startup checkbox not working, as shown in the following screen print:
[image]

onlykey, 2018-11-27:
@timetoseek There is new firmware that addressed the Win7 issue. The version shown is the same though as just the bootloader in the firmware file was updated. Are you able to try loading the firmware again? You can just follow the upgrade guide instructions - https://docs.crp.to/upgradeguide.html

Do you think the app issue you are having is due to the Win7 issue? If the USB device is not recognized when you plug it in the app would be stuck at working.

timetoseek, 2018-11-29:
**Update:** 
I loaded the latest signed firmware (I like the Pac-Man animation!)  
Unfortunately the updated firmware didn't solve the problem... all issues noted above still persist, with Win 7 and with Win 10...    
I just thought of this-- should load the unsigned firmware (via Teensy), as in the past?  Not sure if that makes a difference or not...

onlykey, 2018-11-29:
@timetoseek Yes, as mentioned in the upgradeguide I mentioned you have to update via teensy loader with the .hex file firmware.

timetoseek, 2018-11-29:
**Update:**  New firmware loaded via Teensy and this has resolved all issues with Win 7 & Win 10 BitLocker :)  I will close this issue. 

Please note that the following minor _app-related_ issue persists... it is the Startup checkbox not working, as shown in the following screen print:
> [image]

As it's not Firmware related, I'll close this thread...Thank you!

timetoseek, 2018-11-29:
I opened up a new issue in the OnlyKey-App issues (89)
https://github.com/trustcrypto/OnlyKey-App/issues/89

timetoseek, 2018-12-20:
I am testing OnlyKey with my new laptop, and unfortunately the **BitLocker pre-boot pin** is not working, so I need to reopen this issue. 

This was working just fine with Windows 7 Enterprise, so I closed this issue.  However, for some reason OnlyKey is not working with Windows 10 Enterprise (attached is a screen print of my system info). OnlyKey Firmware is v0.2-beta.7c. 

Below is a video to demonstrate the issue (sorry for the poor quality, and my unsteady hand with the camera :(  But you will get the idea...

[image]

[image]

onlykey, 2018-12-20:
@timetoseek Can you get it to work with any external keyboards?

onlykey, 2018-12-21:
@timetoseek Thanks for testing that. Is there any difference if you already have the OnlyKey plugged in and unlocked when it gets to the bitlocker screen? Is there any difference if you change the typespeed on the OnlyKey?

onlykey, 2018-12-21:
@timetoseek The only other thing I can think of to try is an OnlyKey with the Beta 6 firmware. It may be that there is just a compatibility issue that has always been there but if Beta 6 device works we know that something about Beta 7 is causing the issue.

timetoseek, 2018-12-23:
I tried your suggestion and installed the old Beta 6 firmware. To do so, I had to use the old OnlyKey App Beta 5.0 as required. In any case, this did not solve the issue, please see video below.
[[image]](https://youtu.be/8Km1fQIHFdo
 "BitLocker pre-boot pin with OnlyKey running Beta 6 firmware")

I also managed to update the BIOS on the laptop, thinking that may be a problem. However, the updated BIOS did not resolve the problem.  I also adjusted some of the BIOS settings, but that did not help.  If you think it may be related to a BIOS setting, then please let me know which one(s) and I will adjust.  Screen prints appear below. 
[image]

[image]

[image]

[image]

[image]

timetoseek, 2018-12-23:
During my testing, I also discovered another issue, which I believe may be related to the Windows 10 Enterprise (& BitLocker?) issue described above. When I leave the OnlyKey device plugged in during powerup, it does not work properly during the Windows login screen.  The device is running Beta 7 firmware. The OnlyKey device will work just fine if I plug it in during the Windows login screen (but not before).  A video appears below.
[image]

Has anyone used OnlyKey successfully with Windows 10 Enterprise edition?  If yes are there any settings which need to be modified to make it work?  OnlyKey works perfectly in all other Windows versions I've tried, just not with Win 10 Enterprise with BitLocker.  Actually, I can't turn off BitLocker to see if OnlyKey will work without BitLocker enabled, since I don't have admin access.  If anyone has Win 10 Enterprise with Admin rights, please try it without BitLocker, to see if it works?

onlykey, 2019-01-07:
@timetoseek I am working with our team to see if we can set up Win 10 Enterprise to try and reproduce this issue.

onlykey, 2019-01-09:
@timetoseek So we set up a VM with Win 10 Enterprise and can't reproduce the issue. OnlyKey types as normal at the Bitlocker screen. Here is a video https://youtu.be/Gf6-d2Fu4k0

I know you have already provided a lot of information here but we have not gotten any other reports of issues. Is there any more information you can provide that might help us reproduce this? Have you always experienced the issue on the same type of hardware or is the issue on multiple vendor laptops with Win 7 Enterprise?

timetoseek, 2019-01-11:
> @timetoseek So we set up a VM with Win 10 Enterprise and can't reproduce the issue. OnlyKey types as normal at the Bitlocker screen. Here is a video https://youtu.be/Gf6-d2Fu4k0
> 
Thank you.. I watched the video & looks like it is working perfectly... 

> Is there any more information you can provide that might help us reproduce this? 
> 
I will take another look at all the settings in BIOS and adjust one-by-one (hopefully I won't brick my new laptop)   will post results here in a few days when finished...

> Have you always experienced the issue on the same type of hardware or is the issue on multiple vendor laptops with Win 7 Enterprise?
> 
Works flawlessly on Win 7 Enterprise w/BitLocker installed, no problems at all, I used it for several months on a daily basis on that particular laptop.

juliomaranhao, 2019-02-01:
I have a standard onlykey. Actually I bought 2 (1 reserve) but the first just burned out after many useful months because of a simple stand by. What I learned: don't let the computer do sleep/hibernate then awaike while onlykey is pluged. The key was super hot and no more alive. The USB port is live and OK.

Since then I am using the reserve for more than a year. I tested the last firmwares until yesterday but only beta 5 works. It is the same version I was already using before testing any beta 6 or beta 7x.

I am just writing this here because maybe this #81 issue may not be related to bitlocker at all since I never used bitlocker and I'm also experiencing issues. So don't forget to test previous firmware versions.

P.S.: about Enterprize Readiness, I don't recommend it as the firmware are labeled beta but are actually alpha as new features are added but bugs are not fixed between versions. 
P.S.: I am not trying to fix any more bugs. I am happy with beta 5.

onlykey, 2019-02-02:
@timetoseek Have you seen this issue on just this certain model of HP laptops?

timetoseek, 2019-02-04:
Unfortunately I have not yet been able to find a non-HP laptop running Win 10 Enterprise... all the new laptops I've tested have been the exact same HP model, and all have the same issue.   I will post a note when/if I can find a non-HP laptop running Win 10 Enterprise...

timetoseek, 2019-04-13:
It has been a while, but have not given up trying to figure out this issue :grey_question: :confused:   
today tried this on a different laptop--an older HP EliteBook, running Windows 10 Enterprise. Unfortunately, it is the same problem as the new HP ZBooks...please see video below:
[[image]](https://youtu.be/syRxOj6vzm4
 "Using OnlyKey with BitLocker on an HP EliteBook and Windows 10 Enterprise")

onlykey, 2019-04-15:
@timetoseek Thanks for testing this. I did a bit more research on keyboards not working on bitlocker with HP EliteBook and found this - https://h30434.www3.hp.com/t5/Business-Notebooks/External-keyboard-not-working-in-Bitlocker-screen/td-p/5556449

It suggests that the system may not detect USB devices if fastboot is enabled, are you able to test with this feature disabled in BIOS?

timetoseek, 2019-04-15:
Thank you, yes I tried testing with FastBoot disabled, as the thread suggests, but the same issue persists.  
From the second thread you mentioned, it sounds like a known problem with this laptop :frowning: 

I'm still trying to find a non-HP laptop running Win10 Enterprise, but no luck so far ... Anyway, I've posted a pic/video with today's tests...

Here's the BIOS:
[image]

And here's the video with results:
[[image]](https://youtu.be/TpGfngilWuA
 "Fastboot disabled - testing OnlyKey with BitLocker on an HP ZBook and Windows 10 Enterprise")

onlykey, 2019-04-15:
@timetoseek Since the first thread mentioned that loading too fast is the issue, I wonder if you added a 10 sec startup delay if that would help, or maybe checking USB storage boot.

timetoseek, 2019-04-15:
Just tried with a 10 second boot delay and USB Storage enabled, unfortunately that didn't help.  

:warning: Please note with Win10 Enterprise: OnlyKey must be unlocked _after_ the BitLocker PIN is entered.  Otherwise, it will cause problems on the Windows login screen.  Please view the video below to the very end, where you will see that OnlyKey is sending the last character of my password, in an endless loop, into the input field (the only way to stop it was to physically unplug OnlyKey):
[[image]](https://youtu.be/sVZAAyZQiB8
 "Boot delay and USB Storage Boot enabled - testing OnlyKey with BitLocker on an HP ZBook and Windows 10 Enterprise")

The loop is is the same issue I mentioned in this test video from December, at around video time 1m 20s
https://github.com/trustcrypto/OnlyKey-Firmware/issues/81#issuecomment-449608798

[image]

timetoseek, 2019-10-23:
Having installed the latest OnlyKey firmware [(v0.2-beta.8)](https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v0.2-beta.8 "v0.2-beta.8") as well as BIOS updates from HP and Windows updates, today I retested this issue-- to see if OnlyKey will now work with the BitLocker pre-boot PIN.  To my great delight, OnlyKey now works properly on the pre-boot screen!  Thank you guys for your great work with the firmware :thumbsup:   :clap:

closed 2019-10-23 by timetoseek

onlykey, 2019-10-23:
@timetoseek That is great news! Thanks for retesting it.