trustcrypto/OnlyKey-Firmware #83: teensy wont load firmware

stgarf, 2018-11-30:
I'm trying to upgrade my onlykey from v0.2-beta.5 to v0.2-beta.6 by following https://docs.crp.to/upgradeguide.html (Upgrading Firmware From Beta6 to Beta7). However the teensy loader complains the hex file is too large. My onlykey is plugged in. I verified by opening the onlykey application and unlocking the onlykey.

onlykey, 2018-11-30:
@stgarf That error appears if you don't have the device plugged in before trying to load firmware. You may have to restart teensy loader.

stgarf, 2018-11-30:
Sorry maybe wasn’t clear enough. I’ve tried multiple times. My device is
fine and is plugged in however teensy loader on macOS High Sierra doesn’t
recognize the onlykey—Chrome and chromium-browser both see onlykey just
fine.

onlykey, 2018-11-30:
@stgarf I guess I am still not clear, If your browser sees the key then macOS has to see the key too. Can you try again, this is a common issue when the device is not plugged in before trying to use teensy loader.

stgarf, 2018-11-30:
Yea, I'm sorry that's no longer the case with maybe the newest version of
Teensy loader and the latest available version of macOS High Sierra. The
device is definitely plugged in before opening Teensy loader.

The same problem occurs for me on my LTS 18.04 Ubuntu Chromebook Pixel 2013
using the linux version on Teensy loader.

sheluchin, 2018-12-21:
I can confirm the same issue on Ubuntu 18.04.1 LTS. It's definitely plugged in before trying to load the firmware.

```
14:46:21.358: Teensy Loader 1.40, begin program
14:46:21.383: File "OnlyKey_Beta7_STD_Original.cpp.hex". 143928 bytes, 223% used
14:46:21.383: File contains data beyond chip size, yikes!
14:46:21.383: Listening for remote control on port 3149
14:46:21.383: initialized, showing main window
14:46:47.562: Verbose Info event
```

onlykey, 2018-12-21:
@sheluchin Do you already have Beta 7 firmware loaded? If so then you may have a new OnlyKey model that no longer supports teensy loader. With the new model you load future firmware upgrades through the OnlyKey app - https://docs.crp.to/usersguide.html#loading-onlykey-firmware

stgarf, 2018-12-21:
I'm still having the same problem with Teensy loader and I'm on Beta 5
trying to move to Beta 6 so that I can move to Beta 7.

>

stgarf, 2018-12-21:
> @stgarf <https://github.com/stgarf> If you have beta 5 or 6 you have the
> Teensy bootloader. It is a common issue if you search site:pjrc.com "File
> contains data beyond chip size"
>
> You just go ahead and insert device, load file and bridge the touch points
> to load the firmware. The Teensy loader has to see the device in order to
> know what firmware size is supported.
>
> https://forum.pjrc.com/archive/index.php/t-45213.html

Alright thanks! Got Teensy loader working correctly. For whatever reason it
doesn't work properly on my Ubuntu machine. I've moved to a Mac and managed
to move to Beta 7.

This is probably a separate GitHub issue but now the only problem I've got
is that backup that I've taken from Beta 5 is not restoring properly (even
after restoring my backup phrase initially). The OnlyKey flashes red when
it's in config mode it appears then it seems to reset itself. When I unlock
it, it doesn't contain any backed up data.

Should/Can I run a gpg command across the backup file to verify it locally?

onlykey, 2018-12-21:
@stgarf If its flashing red its encountering an error with the restore. The most likely thing is your backup key used for the restore is not the same as the one used for the backup. Keep in mind PGP keys have multiple subkeys.

stgarf, 2019-01-08:
Thanks for the guidance here. Firmware updated but the restore will not work regardless 👎 No big problem to set everything back up though.

closed 2019-01-08 by stgarf

sheluchin, 2019-02-14:
@onlykey I'm on Firmware v0.2-beta.6c. Your link indicates that I should use the instructions [here](https://docs.crp.to/upgradeguide.html) to upgrade, but I get stuck because Teensy tells me `OnlyKey_Beta7_STD_Original.cpp.hex (too large!)`.

hacksingh, 2019-02-19:
Hi, 
I am still not able to get pass the teensy uploader v 1.40 and gettingsize error. I am on version v0.2-beta.6. I tried plugging in the key multiple times. 

11:14:59.917: Auto Button event
11:14:59.921: Auto mode: enabled
11:16:59.583: Open File event
11:17:01.828: File "OnlyKey_Beta7_IN_TRVL_Color.cpp.hex". 113116 bytes, 175% used
11:17:01.831: File contains data beyond chip size, yikes!

stgarf, 2019-02-19:
@hacksingh I got this working by basically ignoring the warning and bridging the two points on the onlykey to flash the firmware. Everything appeared to work correctly for me...

hacksingh, 2019-02-19:
> bridging the two points on the onlykey to flash the firmware

Thanks for the reply. Can you please elaborate on how to perform this step, "bridging the two points on the onlykey to flash the firmware"?

onlykey, 2019-02-19:
@hacksingh https://docs.crp.to/upgradeguide.html