trustcrypto/OnlyKey-Firmware #85: Request Dvorak layout

chaomodus, 2019-01-09:
It'd be brilliant if a basic dvorak layout could be supported (and/or we could specify a layout file externally). The app interacts correctly with the device but it prints out, obviously, in the QWERTY layout. 

Thanks!

simeonjmcg, 2019-03-29:
I would like to second this. It's extremely inconvenient to have to switch keyboard layouts every single time I type in a password. I would very much appreciate if we could get simple Dvorak layout, much like we have international keyboard layouts.

toastal, 2019-04-15:
Since I use Dvorak at the software level, I've been typing my data as if it's QWERTY and Dvorak will come out. But this means I have to have Dvorak set on other machines as well, like my phone. This would be a nice addition.

onlykey, 2019-04-24:
We have been a bit busy with other features but we do want to support additional keyboard layouts. Would you be able to do a pull request for this? Here is an example of a pull request to add a keyboard layout - https://github.com/trustcrypto/OnlyKey-Firmware/pull/90

simeonjmcg, 2019-04-25:
Yeah, I understand that. I worked on adding support manually, but I got stuck at the build process. I loaded it into the Arduino IDE, but I then realized that I didn't have the right dependencies. Could I have build instructions so I can test it? I prefer to only PR code that I've tested.

onlykey, 2019-04-26:
@simeonjmcg Here are some build instructions. Let me know if these work for you. https://github.com/trustcrypto/OnlyKey-Firmware/issues/59

simeonjmcg, 2019-06-11:
@onlykey I was able to successfully build and flash the developer OnlyKey, but when I plug it in, it prompts for a pin instead of prompting setup. Is there an initial password built into the dev build?

onlykey, 2019-06-12:
@simeonjmcg Yes, for easy development the device is auto-configured. 
Pin - 1111111
2nd pin - 222222
SD pin - 3333333
passphrase - aaaaaaaaaaaaaaaaaaaaaaaaa

eternaltyro, 2019-12-13:
There's now a PR for Dvorak added 20 days or so ago. I don't know enough C to review it, but at a high level it looks good.

onlykey, 2020-01-06:
@eternaltyro @simeonjmcg Do either of you have the ability to test Dvorak support if I send you an OnlyKey loaded with firmware supporting this? Ive never used a Dvorak keyboard so it would be best to have someone familiar test this. Let me know and if so shoot an email to support <at> crp <dot> to

simeonjmcg, 2020-01-06:
@onlykey a while back you sent me a dev board to work on developing this. I got close to what I think would work, but never got a chance to test it because, even without my changes, the board wouldn't work right after I flashed it to the device. I can't say for sure if it's actually a bug or something I didn't set up correctly, but I quickly got busy with other things that unfortunately I forgot about it. The board is still working, it just has less-than-working firmware on it right now.

If you could get me a prebuilt image that I can flash onto the dev board, I should be able to test this.

onlykey, 2020-01-06:
@simeonjmcg OK, I should be able to get something out by next week. To flash to the dev board you have to use Teesny loader and the instructions here - https://docs.crp.to/legacyupgradeguide.html#loading-onlykey-firmware with your custom firmware

atoponce, 2020-07-29:
I'm looking for a U2F hardware token that supports the Dvorak layout. Any updates on this?

onlykey, 2020-07-30:
@atoponce U2F does not use keyboard layout, for things like password management keyboard layouts are used.

onlykey, 2020-10-09:
I am going to add support for Dvorak so we can at least test this in the next release. I am thinking we might have to add multiple dvorak layouts to fully support this, i.e. dvorak on US keyboard, dvorak on US International keyboard, etc. For right now I would like to just test this on US keyboard. Please review changes and let me know if I missed anything or if any of the key mappings are incorrect - https://github.com/trustcrypto/OnlyKey-Firmware/commit/dcad83a6d9616c9c5ef7bde09192f73f66a33fd6

eternaltyro, 2020-10-10:
@onlykey will you also be releasing an alpha with this change so I can update my hardware and test this? My hardware is lying unused for the moment.

simeonjmcg, 2020-10-10:
@onlykey Glad to see some movement on this. Back when I tried working on this I was unable to activate the mapping once I wrote it, even with the command line application. Has this been tested on the hardware? If so, I can help work on expanding the supported variants.

One use case that would work really well for me is to have layout overrides for specific slots. This would be useful in cases where you use different layouts interchangeably. I don't often use QWERTY, but when I do, it would be useful to have that mapping on hand in another slot.

@eternaltyro I have been using the onlykey with Dvorak by simply manually mapping passwords backwards, eg. putting in "Ra;;,aoh123" into the password field for "Password123" in US dvorak. This also allows me to have one slot for Dvorak and one for QWERTY.

closed 2020-11-20 by onlykey