trustcrypto/OnlyKey-Firmware #89: Cannot compile beta 7

takd, 2019-03-31:
I am on Ubuntu 18.04 LTS. 

I am trying to add a new keyboard layout to the firmware - so I did everything according to issue #59, but failed to install my compiled firmware. I did get into the bootloader mode by bridging the pins, but the Teensy loader always had a problem with the firmware size, and nothing happened.

After looking more deeply into the issue, I realized that I have firmware v0.2-beta.7c installed, which according to [this comment](https://github.com/trustcrypto/OnlyKey-Firmware/issues/83#issuecomment-449413050) means that I cannot load the firmware with the Teensy loader. I tried installing it from the OK desktop app, but I get "Error Firmware Block too large." The hex file is 548,4kB, which seems different than the one under Releases, but I didn't change anything in the downloaded .ino file.

What should I do then to correctly compile and upload the self-compiled frimware?

takd, 2019-03-31:
Update: I had an idea, and entered bootloader mode with the OK app open, instead of the standard update process with config mode. I succeeded in entering bootloader mode, which the app recognized, but I still got the same "Error Firmware Block too large." error message, so the update was unsuccessful.

onlykey, 2019-04-01:
@takd You can only load signed firmware through the OnlyKey app. Also you would not load the HEX file through the app that format is for Teensy loader. If you have an older model onlykey there is a hardware bootloader (2nd smaller chip can be seen) this is used with the Teensy loader app. If you have the hardware bootloader you can load your own firmware with teensy loader and you can ignore size warnings. If you don't have the hardware bootloader you will have to get a developer onlykey with one. Its not available on our store you will have to send an email to admin at crp.to to request one.

takd, 2019-04-01:
Is the second smaller chip labeled something like this?
>16000
>K733 Y
This is the only component on mine that looks like a small chip, but I couldn't find a datasheet at first search.

onlykey, 2019-04-01:
@takd 
[image]

The numbers are a bit small to read but see the blue arrow. If it has this chip this is the hardware bootloader and you just need to use Teensy loader to load firmware. Load your firmware, make sure teensy loader has the green auto button clicked, bridge the touchpoints.

takd, 2019-04-01:
Well nope, I don't have a hardware bootloader then. I will probably try to add the keyboard layout without testing, and post it at #79, or send a pull request. Thanks for the assistance though!

onlykey, 2019-04-01:
@takd Send me an email t at crp.to we will figure something out.

closed 2019-05-01 by takd