trustcrypto/OnlyKey-Firmware #94: Release files:  IN_TRVL vs STD?

TopView, 2019-10-28:
What is the difference between [Release notes](https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v0.2-beta.8)  'assets':

Signed_OnlyKey_Beta8_IN_TRVL_Color.txt
Signed_OnlyKey_Beta8_STD_Color.txt

What does the IN_TRVL and STD stand for.  For that matter what is "Color" all about?

Thanks

phrogg, 2019-10-28:
Color is the normal stick with the colored LED. An Int Travel is the international travel stick, without the color LED but with a Label called LED.

closed 2019-10-29 by TopView

onlykey, 2019-10-29:
@TopView The original OnlyKey that was discontinued a couple of years ago did not have a color LED. We stuck with the naming convention but don't really need the word color anymore.