trustcrypto/OnlyKey-Firmware #98: HMAC Challenge Response Leads to Unresponsive OnlyKey

schlomie, 2019-11-25:
Per our discussion [here](https://github.com/trustcrypto/trustcrypto.github.io/commit/2583481278cd5cda2931db6910fbc3d675321478#r36077348), this issue describes the behavior I am seeing on the OnlyKey when using the HMAC Challenge Response to open a kdbx file with KeePassXC.

Firmware: v0.2-beta.8c
App: v5.2.0

Per your instructions, I added my 20 byte  HMAC secret (generated with the ykper tool) to ECC Slot 30 on the Advance Tab of the app, padding the right with zeros to fully supply the required 32 bytes.

The OnlyKey was disabled (became unresponsive to password touches<sup>1</sup>) in exactly two ways, although I think both are triggered by the same mechanism.

1. Clicking the Refresh Button for the Hardware Key input in KeePassXC
2. Pressing any button on the OK when prompted (for the Challenge/Response) by KeePassXC.

Steps to reproduce 1.

1. Have your YK HMAC Secret Loaded
2. Insert and Unlock the OnlyKey
3. In some text editor/entry/whatever test the password type capability by pressing a slot on OK
4. Without a Yubikey inserted<sup>2</sup>, start KeePassXC and open a database.
5. Click on Refresh next to the Hardware Key entry
6. **Repeat step 3 - The OK no longer types out passwords**
7. If you type in the password for the kdbx and attempt to open the file, KeePassXC attempts the Challenge Response against the OK (the LED flashes RED until pressed) and the kdbx is successfully opened.
8. Locking the kdbx and repeating step 7 still successfully prompts the OK for HMAC.

Steps to reproduce 2.

1.  After unlocking (step 7 above) re-lock the kdbx file - KeePassXC will be in its waiting state, prompting you to _Unlocked KeePassXC Database_
2. Note the Hardware Key entry will show **OnlyKey[ser#] Challenge Response - Slot 1 - Press**
3. Disconnect, Re-Insert and Re-Unlock the OK
4. Repeat step 3 above, proving passwords are typed
5. Type the password in KeePassXC and unlock the file
6. You are again prompted to press the OK and the LED flashes RED again.
7. Pressing any button on the OK successfully responds to KeePassXC and the file is opened.
8. Repeat step 3 above - **Once again, the OK no longer types out passwords<sup>1</sup>**

I will upload some screenshots/video demonstrating this when I have a bit more time.

---
<sup>1</sup> This HMAC soft lock (as we might call it) only affects using the OnlyKey to type out passwords.  **onlykey-agent still works**, after OK has become unresponsive.  Although interestingly, in this state, an extra `<Enter>` press is necessary with onlykey-agent.   Under normal operating conditions, the moment the 3-digit challenge is met (or pressed when 3-digit challenge is disabled) the secure shell session is opened.  After the HMAC soft lock as described above, the moment the 3-digit challenge is met, the terminal will wait until `<Enter>` is pressed on the keyboard.

<sup>2</sup> I've noticed KeePassXC favors the Yubikey.  If the YK is inserted prior to all of this, the OK isn't even considered.

Additionally (and possibly related,) [this](https://github.com/trustcrypto/onlykey-agent/issues/3#issue-330740967) is _kind of_ still an issue (I need to update that issue with new findings post beta8.)  Any time the OnlyKey is used for any operation outside of typing passwords, the OnlyKey App no longer detects the OK.  The OK still functions correctly, types out passwords (sans HMAC soft lock,) works with onlykey-cli and onlykey-agent and functions correctly as a U2F token.   To get the App to recognize the OK again, it must be disconnected, re-inserted and re-unlocked.

tafli, 2020-03-09:
I can reproduce this type of unresponsiveness on my system as well:

OS: Manjaro Linux (with Kernel v5.5.7)
Firmware: v0.2-beta.8c
App: v5.2.0
KeePassXC: v2.5.3

@schlomie: I guess, you're also on a Linux based system?

Checked it on MacOS, it works as expected and OK does type passwords after having unlocked database with KeePassXC.

schlomie, 2020-03-09:
@tafli Yes.  On Void, 5.4.24

nkichukov, 2020-03-18:
Same here, GNU/Gentoo Linux with kernel 5.4.13. Apps are same versions as @tafli reported.
ykpers is 1.20.0

nkichukov, 2020-03-18:
Further investigation on this issue revealed that whenever the OK is detected by keepassxc for challenge response, it is actually removed from the kernel (unbind). This is why it works no more as a keyboard, it simply does not exist and the OK does not present itself afterwards (after being listed or used as challenge response factor) as HID device to the kernel. Here is what the udevadm monitor shows:

OK added to computer:

```KERNEL[2343.716367] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
KERNEL[2343.717796] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B (hid)
KERNEL[2343.717832] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23 (input)
KERNEL[2343.777729] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::numlock (leds)
KERNEL[2343.777789] change   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::numlock (leds)
KERNEL[2343.777813] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::capslock (leds)
KERNEL[2343.777824] change   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::capslock (leds)
KERNEL[2343.777834] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::scrolllock (leds)
KERNEL[2343.777845] change   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::scrolllock (leds)
KERNEL[2343.777854] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::compose (leds)
KERNEL[2343.777914] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::kana (leds)
KERNEL[2343.777927] change   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::kana (leds)
KERNEL[2343.777974] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/event5 (input)
KERNEL[2343.778124] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/hidraw/hidraw0 (hidraw)
KERNEL[2343.778141] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B (hid)
KERNEL[2343.778155] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
KERNEL[2343.778167] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.1 (usb)
KERNEL[2343.779027] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.1/0003:1D50:60FC.000C (hid)
KERNEL[2343.779150] add      /class/usbmisc (class)
KERNEL[2343.779167] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.1/usbmisc/hiddev0 (usbmisc)
KERNEL[2343.779238] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.1/0003:1D50:60FC.000C/hidraw/hidraw1 (hidraw)
KERNEL[2343.779256] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.1/0003:1D50:60FC.000C (hid)
KERNEL[2343.779270] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.1 (usb)
KERNEL[2343.779283] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2 (usb)
KERNEL[2343.779967] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/0003:1D50:60FC.000D (hid)
UDEV  [2343.780110] add      /class/usbmisc (class)
KERNEL[2343.780256] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/usbmisc/hiddev1 (usbmisc)
KERNEL[2343.780352] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/0003:1D50:60FC.000D/hidraw/hidraw2 (hidraw)
KERNEL[2343.780394] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/0003:1D50:60FC.000D (hid)
KERNEL[2343.780420] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2 (usb)
KERNEL[2343.780482] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1 (usb)
UDEV  [2343.851169] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2 (usb)
UDEV  [2343.854682] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/usbmisc/hiddev1 (usbmisc)
UDEV  [2343.854710] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/0003:1D50:60FC.000D (hid)
UDEV  [2343.859022] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/0003:1D50:60FC.000D/hidraw/hidraw2 (hidraw)
UDEV  [2343.859542] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2/0003:1D50:60FC.000D (hid)
UDEV  [2343.860215] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.2 (usb)
UDEV  [2347.751246] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
UDEV  [2347.752892] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B (hid)
UDEV  [2347.754825] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/hidraw/hidraw0 (hidraw)
UDEV  [2347.754896] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23 (input)
UDEV  [2347.756032] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::numlock (leds)
UDEV  [2347.756126] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1D50:60FC.000B/input/input23/input23::scrolllock (leds)
[... cut ...]
So a simple workaround to continue using your OK without having to remove it from the computer thus having to type your unlock code again is to simply 'bind' to the usb subsystem manually, ie:

`echo "2-1.1" > /sys/bus/usb/drivers/usb/bind`

Because the OK was never power-cycled, it keeps your profile open and everything continues to just work.

After the device got detected or used for challenge response it needs to tell the kernel that it is again a hid(hidraw) device so it can be bound and used again to type.

Perhaps this can become in-firmware fix or udev-rule type of thing.
HTH,
-Nik

nkichukov, 2020-03-19:
one correction, you'd workaround with binding the whole USB device(as initially suggested) if you have previously unbound it, which is not the case when OK is detected by keepassxc. The detection only removes/unbinds the HID device. Thus the correct 'bind' should go to usbhid instead, something like this udev rule:

```ACTION=="unbind", SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_interface", ENV{INTERFACE}=="3/1/1", ENV{PRODUCT}=="1d50/60fc/100", ENV{DEVTYPE}=="usb_interface", RUN+="/bin/sh -c '/bin/echo -n $kernel > /sys/bus/usb/drivers/usbhid/bind'"```

However this most likely prevents the 'press any key to approve the challenge response' functionality... which essentially breaks the whole challenge response concept.

In addition, I see strange misbehaviour where the device sporadically disappears from the list of devices for challenge response...

nkichukov, 2020-03-19:
Seems very related to:
https://github.com/trustcrypto/OnlyKey-Firmware/issues/107
https://github.com/keepassxreboot/keepassxc/issues/4400

onlykey, 2020-03-20:
@nkichukov Excellent job here tracking down the issue. I think with this info I will have what I need to find a solution, time to dust off the USB protocol analyzer.

onlykey, 2020-04-02:
@nkichukov I did some testing today with Yubikey to see if it has the same issue, looks like it does. I set a static password in slot 1 (keyboard), then HMAC SHA1 in slot 2. Keyboard works fine until I attach the device to KeepassXC, after attaching it no longer works as a keyboard. Going to do some more testing to see what options are available, in firmware we might be able to do a time based reconnect USB 10 seconds after receiving last HMAC related event to reconnect key to the host. I am thinking the best fix would probably be if this is just an issue that could be fixed in KeepassXC so I will look into that too.

nkichukov, 2020-04-03:
Hi Tim,

As to the short-term workaround, at least applicable on linux, the additional udev rule I posted before I have modified a bit more to include a delay, so instead of doing it at the firmware level, I do it at the OS level:

```ACTION=="unbind", SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_interface", ENV{INTERFACE}=="3/1/1", ENV{PRODUCT}=="1d50/60fc/100", ENV{DEVTYPE}=="usb_interface", RUN+="/bin/sh -c 'sleep 5 && echo I ran in usb system $kernel, $parent >> /tmp/udevrule && /bin/echo -n $kernel > /sys/bus/usb/drivers/usbhid/bind'"```

What this does is, every time I get to the unlock keepassxc database screen, the onlykey does not work for 5 seconds to type my password, then it works as keyboard again, then I can select the challenge response slot and follow the process. Not ideal, but temporary resolves the problem.

Why the challenge response slot enumeration function disables the keyboard? Why is this needed? Can this be avoided? The listing of available slots should not disable the keyboard feature, the actual response generation(when opening the database) should as it prompts for a key to be pressed, without generating any output. So instead of adding 10 seconds delay in firmware to reattach, we should perhaps see if we can completely avoid this scenario(in keepassxc, yubikey-personalization as keepassxc relies on it and OK-firmware)?

onlykey, 2020-04-14:
@schlomie @tafli KeepassXC has pushed a fix that addresses hmacsha1 issues, including the issue where you can't use OnlyKey as a keyboard after using with KeepassXC on Linux. Are you guys able to test it out and see if you have any issues?

https://github.com/keepassxreboot/keepassxc/pull/4584

I have tried it out and this seems to fix the issue, @nkichukov tested as well.
```
git clone https://github.com/keepassxreboot/keepassxc -b develop
cd keepassxc/
git checkout 048386c3f9a655fb19d9eff6b52b7fd85b64a972

build instructions here - https://github.com/keepassxreboot/keepassxc/wiki/Building-KeePassXC
```

schlomie, 2020-04-14:
I built it from that commit, still the same issue with the OnlyKey - AND it no longer detects the Yubikey.

Debug Info:

> KeePassXC - Version 2.6.0-snapshot
> Build Type: Snapshot
> Revision: 048386c
> 
> Qt 5.14.2
> Debugging mode is disabled.
> 
> Operating system: void
> CPU architecture: x86_64
> Kernel: linux 5.5.16_1
> 
> Enabled extensions:
> - Auto-Type
> - Browser Integration
> - SSH Agent
> - KeeShare (signed and unsigned sharing)
> - YubiKey
> - Secret Service Integration
> 
> Cryptographic libraries:
>  libgcrypt 1.8.5

onlykey, 2020-04-14:
@schlomie Hmm I am not seeing that, I know @nkichukov mentioned in the keepassXC issue that it was now working for him as well.

KeePassXC - Version 2.6.0-snapshot
Build Type: Snapshot
Revision: 048386c

Qt 5.12.5
Debugging mode is disabled.

Operating system: Kali GNU/Linux Rolling
CPU architecture: x86_64
Kernel: linux 5.4.0-kali4-amd64

Enabled extensions:
- Auto-Type
- Browser Integration
- SSH Agent
- KeeShare (signed and unsigned sharing)
- YubiKey
- Secret Service Integration

Cryptographic libraries:
 libgcrypt 1.8.5

I do see that by default Hardware Key just says select hardware key, you have to click the arrow to see the Yubikey
[image]

[image]

When both Yubikey and OnlyKey are inserted I see that the OnlyKey slot assigned to my KBDX is autoselected

[image]

[image]

At this point I can open a text editor and select one the static passwords in my OnlyKey and it now types. I tested this on Debian.

schlomie, 2020-04-14:
Interesting.   Yeah, I have both my Yubikey and Onlykey plugged in, and the only option available is the OnlyKey

[image]

But anyway, I'm still seeing the same behavior, where the OnlyKey "locks up" (so to speak,) where it no longer types out passwords.  All other functions continue to work (including the HMAC-SHA1 challenge response on keepassxc)

nkichukov, 2020-04-15:
This has been resolved for me, however I am at commit a430c304e2291f25794ae22772036742743dd689 and you are at the next one.

I do not have yubikey to test, but onlykey works as expected.

After I recompiled with the patches included, I did see some strange behaviour, but next day after system reboot everything was working as expected(perhaps old preserved libraries were loaded).

I do like the new functionality, where keepassxc 'remembers' if your database file makes use of hardware key and only lists/refreshes the tokens and then pre-selects those for the databases that actually make use of it.

Typing from Onlykey works after keepassxc token refresh without any additional udev rules required now.

I do not like their(keepassxc) new theme, but functionality-wise: all just works.

onlykey, 2020-04-15:
@schlomie I don't see the "WARNING" message in your screenshot, that should appear if you are running Keepassxc built from the commit 048386c3f9a655fb19d9eff6b52b7fd85b64a972

schlomie, 2020-04-15:
Bah!   I dismissed the warning before snapping the pic.

[image]

Here's the debug, indicating it is built from that commit.

> KeePassXC - Version 2.6.0-snapshot
> Build Type: Snapshot
> Revision: 048386c
>
> Qt 5.14.2
> Debugging mode is disabled.
>
> Operating system: void
> CPU architecture: x86_64
> Kernel: linux 5.5.16_1
>
> Enabled [...]

schlomie, 2020-06-08:
So I've built KeepassXC a few times now, since [this PR](https://github.com/keepassxreboot/keepassxc/pull/4584) was merged, and once again last night with the 2.6.0-beta1 tag.

The changes made to [#4584](https://github.com/keepassxreboot/keepassxc/pull/4584) before it was merged made it work without issue with OnlyKey.   Interestingly though, it no longer works with my Yubikey 4 or 5.

**Yubikey**
[image]

**OnlyKey**
[image]

I thought it might be an issue with my build environment, so I checked out the 2.5.4 tag and built it using the same libs and toolchain - result was the same as the pre-built 2.5.4 binaries.  Yubikeys would work fine, OnlyKey would not.

**Anyone else want to try building the 2.6.0-beta1 tag and test with both the OK and YKs?**

The 2.6.0 milestone only has a few outstanding issues remaining, so they are likely only a few days or so away from a release.   I think I'll wait until after before I open a bug report over there.

> KeePassXC - Version 2.6.0-beta1
> Build Type: PreRelease
> Revision: e5b0219
> 
> Qt 5.15.0
> Debugging mode is disabled.
> 
> Operating system: Artix Linux
> CPU architecture: x86_64
> Kernel: linux 5.6.14-artix1-1
> 
> Enabled extensions:
> - Auto-Type
> - Browser Integration
> - SSH Agent
> - KeeShare (signed and unsigned sharing)
> - YubiKey
> - Secret Service Integration
> 
> Cryptographic libraries:
>  libgcrypt 1.8.5

Also - now built on Artix (arch) instead of Void - just to rule out any irregularities.

schlomie, 2020-07-16:
With 2.6.0 out the door, I guess I can comment on this further.   Looks like I wasn't the only one who encountered the issue of Yubikey not being detected:

https://github.com/keepassxreboot/keepassxc/issues/4987

But it was a pretty simple fix and has been tagged for 2.6.1:

https://github.com/keepassxreboot/keepassxc/pull/5004

Anyway - nevermind all that.   **OnlyKey works well with 2.6.0** and I no longer feel this issue needs to be open any longer.

closed 2020-07-16 by schlomie