trustcrypto/OnlyKey-Firmware #99: Assuming the keyboard layout of the operating system

eternaltyro, 2019-12-13:
I would like for OnlyKey to assume the current active keyboard layout of the operating system so I don't have to explicitly switch between layouts for OnlyKey to work. This would be a nice addition since I know many people who use alt-layouts (Dvorak, Colemak) to whom I want to recommend OnlyKey.

In fact, I was quite flummoxed when the initial instructions were output in quite a weird format (because I had Dvorak setup) when I tried setting it up first. It's encouraging that pull requests are beginning to come-in for alt-layouts, but auto-switching could help a lot in adoption.

eternaltyro, 2019-12-13:
FWIW, I use `setxkbmap dvorak` and `setxkbmap us` to switch between layouts

onlykey, 2019-12-13:
@eternaltyro This is possible, it would require an agent to run on the computer thought to inform the OnlyKey of what keyboard profile is in use.