trustcrypto/OnlyKey-Firmware #103: Webcrypt from other domains

xeor, 2020-01-19:
It sais in the docs `The OnlyKey currently only works with apps.crp.to.`, which is good for phising, but very bad for those who wants to host everything themself.

I get that the apps are verifiable, domains under https, and it only communicates to the onlykey. But I can't verify this every time I use this page.. What if it is hacked? What if I want to brand it in company colors or do some changes?

The restriction is good, but it should absolutely be configurable, so that I can use my own domain.

Is this in the pipeline?

onlykey, 2020-01-20:
@xeor Its not currently in the pipeline. As you mentioned if we allowed any domain then it would be easier to phish users to say decrypt a message/file. We would need to add a feature to support custom domains that allowed a company to add their own domains. If there was a large demand for this like a company wanted to rebrand and was purchasing a significant quantity of devices we would certainly be able to do this.

xeor, 2020-01-20:
Allowing any domains would not fly, agree.. To be honest, I almost didn't buy them because I read beforehand that the domain was locked like this. For my own sake, I won't be able to use this domain without worrying, so I would probably end up not using it at all :(

Crossing my fingers that this can be implemented without a big company buying a lot of hardware.. Maybe this issue even make some companies desiring not to buy...?

onlykey, 2020-01-21:
@xeor Keep in mind, just wanted to clarify the feature we are talking about here is OnlyKey WebCrypt. There is no restriction on only using OnlyKey's other features like password management, FIDO U2F, FIDO2, Challenge-response etc. The restriction is you can only use the OnlyKey WebCrypt app with official app which is at domain apps.crp.to.

xeor, 2020-01-21:
Absolutt.. This is only above the webapps. I like it idea of using the u2f api for 2way community will the key like this, but it would be nice to be able to customize/bring-your-own functionality

onlykey, 2020-01-21:
@xeor If someone is looking to build a web app that would be something a good portion of our user's might benefit from we would consider whitelisting a domain for this app in a future firmware release.

xeor, 2020-02-03:
Sorry for the late response..

But for an organisation to be able to the web-apps, they need to at least be hosted internally.
For myself as well, I really want to host this myself on my own domain.

A setting like that needs to be configured using the app, so is there really a big risk? What if can only be configured during the initial setup?

Or maybe just add another domain as well that can't be public? Like `webcrypt.local` or just `webcrypt`?

xeor, 2020-05-15:
Has there been changes in this manner? I am hesitate to use my only-keys till there is a way to do this..

onlykey, 2020-05-17:
@xeor No, there isn't any plans to do this right now. This would not be a small feature or an easy to implement change.

alexhk90, 2024-12-27:
As an alternative to running from a separate domain, or hosting privately, would it be possible to run the OnlyKey web apps ("https://apps.crp.to/") offline, for example as a Progressive Web App?