trustcrypto/OnlyKey-Firmware #104: LED doesn't work with OnlyKey Original on BETA-8

yaDMdivaD, 2020-01-24:
I have 3 OnlyKeys now and love them.  I use my original only key to let friends test it out before they buy their own.  I recently attempted to put beta-8 on the original and while it does work, the LED now does not work. 

Is there a fix for this or will a new version of beta-8 be released just for the original?

onlykey, 2020-01-24:
@yaDMdivaD We discontinued the original back in 2016, we continued to support it for up to beta7 but that was the last release supporting the original.

yaDMdivaD, 2020-01-24:
I kind of assumed that but wasn't sure cause i saw the original referenced in the source code for 8.  No worries. Thank you for the quick reply.

closed 2020-01-24 by yaDMdivaD