trustcrypto/OnlyKey-Firmware #105: Keyboard typing speed per PIN

yaDMdivaD, 2020-01-24:
I suppose this is a feature request not an issue.

When using the OnlyKey to store BitLocker recovery keys, I have to set the keyboard speed to 3 or less for it to allow me to type the recovery key.  Would it be possible in a future release to be able to assign a speed preference per primary / secondary PIN?  This way my normal everyday use slots can be speedy but the BitLocker keys on be on the secondary PIN and set to a speed of 3.

onlykey, 2021-11-26:
Will add this feature in firmware v2.1.3

llajas, 2022-06-27:
I’m considering purchasing the device, but I have a serious need for slower type speed on specific slots while maintaining normal/fast speed on others slots. Anticipating the release of the next firmware which will likely fit my exact needs! 😊