trustcrypto/OnlyKey-Firmware #118: Configuring HMAC-SHA1 getting "No Hardware keys detected" using Onlykey v0.2-beta.8c, Keepass 2.6.2,

CZoll, 2020-11-29:
I am getting "No Hardware keys detected". The instructions are a little vague for using HMAC-SHA1 for KeepassXC but I followed them the best I could. The Onlykey was connected directly to motherboard USB port.

**Steps followed**
1) Cleared slot 1 on the Onlykey
2) Enabled either Yubikey OTP, or UF2 via OnlyKey Slot 1a Configuration
2) Created a new db in KeepassXC 
3) General DB Information name="testing HMAC" <continue>
4) Defaults <continue>
5) Entered password and confirmed
6) Select "Add additional protection"
7) Scroll down to YubiKey Challenge-Response
8) Selected Refresh "No hardware keys detected"

**Keepass info**
KeePassXC - Version 2.6.2
Revision: e9b9582

Qt 5.15.1
Debugging mode is disabled.

Operating system: Windows 10 Version 1909
CPU architecture: x86_64
Kernel: winnt 10.0.18363

Enabled extensions:
- Auto-Type
- Browser Integration
- SSH Agent
- KeeShare (signed and unsigned sharing)
- YubiKey

Cryptographic libraries:
 libgcrypt 1.8.6

**Onlykey information**
App v5.2.0
Firmware v0.2-beta.8c

**Windows information** 
OS Name	Microsoft Windows 10 Pro
Version	10.0.18363 Build 18363

**Hardware**
BaseBoard Manufacturer	ASRock
BaseBoard Product	Z77 Extreme9
Etron USB 3.0 eXtensible Host Controller - 1.0 (Microsoft)	PCI\VEN_1B6F&DEV_7052&SUBSYS_70521849&REV_00\6&43FAD4F&0&003800E7
Etron USB 3.0 eXtensible Host Controller - 1.0 (Microsoft)	PCI\VEN_1B6F&DEV_7052&SUBSYS_70521849&REV_00\6&27CF0CAE&0&004000E7
Intel(R) 7 Series/C216 Chipset Family USB Enhanced Host Controller - 1E26	PCI\VEN_8086&DEV_1E26&SUBSYS_1E261849&REV_04\3&11583659&0&E8
Intel(R) 7 Series/C216 Chipset Family USB Enhanced Host Controller - 1E2D	PCI\VEN_8086&DEV_1E2D&SUBSYS_1E2D1849&REV_04\3&11583659&0&D0
Intel(R) USB 3.0 eXtensible Host Controller - 1.0 (Microsoft)	PCI\VEN_8086&DEV_1E31&SUBSYS_1E311849&REV_04\3&11583659&0&A0

onlykey, 2021-11-26:
Following up on old issue, any recent issues seems everything works well with latest KeePassXC and OnlyKey firmware

CZoll, 2021-11-30:
> "No Hardware keys detected" using Onlykey v0.2-beta.8c, Keepass 2.6.2,
> (#118)

I am not sure I am the person you meant to send this to

<https://github.com/trustcrypto/OnlyKey-Firmware/issues/118#issuecomment-980
6IQVANCNFSM4UGW3CKA> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/AR54FAX7HXBDFCIEDHY2YHLUN6IQVA5CNFS
M4UGW3CKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHJU26NQ
.gif>

closed 2021-11-30 by onlykey