trustcrypto/OnlyKey-Firmware #119: Changing Backup Key taints Yubio OTP

jenszo, 2021-01-23:
I just experienced strange behaviour with my set of onlykeys which I relate to the firmware, yet I'm not sure.
As a matter of fact, I just locked myself out of a (financial) service for which I am using the Yubico OTP as a second factor.

It appears that my OnlyKey suddenly keeps producing the same Yubico OTP over and over again. https://demo.yubico.com made me aware of that: "REPLAYED_OTP" The good thing is, I made a backup just a day ago which is working. U2F and Google OTP still work as expected on both OnlyKeys.

**What happened:**
* I have two OnlyKeys, one of which is a regularly updated clone of my first one.
* According to the docs, if a slot is already set, importing a backup only overwrites the password, not the username => also, it appears, it also does not update any OTP changes I made in the past; neither Yubico nor Google OTP. Slightly shocked, that my backup key did not work with any of the 2F services I subscribe to, I made it to self destruct to re-imported a full backup.
* I also decided to change my backup passphrase to a more secure one, did so and updated it on the first OnlyKey and did a fresh backup.
* Setting up the **second** OnlyKey with the new passphrase, importing the backup => works as expected on all OTP secured services.
* Since then, however, the **first** OnlyKey keeps producing the inconsistent Yubico OTPs.

This leads me to the impression, that changing the backup passphrase, somehow tainted the Yubico security features. Other OTP mechanisms are unaffected.

Can anyone confirm this or did I miss something?

OnlyKey App: 5.3.0 on Manjaro Linux Kernel 5.10
Firmware:       0.2-beta.8c

jenszo, 2021-01-23:
After trying to reset and restore the **first** OnlyKey with the existing backup, Yubico OTP still results in "REPLAYED_OTP".
Digging into the result and how Yubico OTP works, I found it uses a counter unlike Goolgle OTP which uses a timestamp (if I'm not mistaken). **duh!**

So I was preemptively judging the backup key for no reason. So the issue is that there is no fixable issue other than the user itself, really. My bad.

However, I could not find anything related in the documentation. This, however, is crucial to the less tech-savy (and the users like me who do not sufficient research into it) to know. 

Current documentation is leading to the assumption to be able to create full-featured clones of OnlyKeys to be used in the event of one getting lost.

To solve the backup issue for the user, a second OnlyKey must use its own Yubico Identity and be registered as some sort master key / secondayr 2F to any of the services used.

onlykey, 2021-01-25:
@jenszo Yes, as you found you can only use one OnlyKey or Yubikey with the same Yubico OTP key. The counter will become out of sync on whichever device was not used last. We do have some mention in the docs but this could be better:

'Pro Tip: Keep in mind that once you write this configuration to OnlyKey you can no longer use a Yubikey with the same configuration. Attempting to do this causes one of the devices to be out of sync.'

We are in future release going to support 24 Yubico OTP slots the same as 24 TOTP are supported so that would allow having multiple keys assigned to an account.

jenszo, 2021-02-04:
Sounds fantastic. Thanks for getting back to this ticket.
Feel free to close to your convenience.

closed 2021-02-05 by onlykey