trustcrypto/OnlyKey-Firmware #122: Slot's data write protection with touch

ealfie, 2021-04-20:
Hi all!

I find it really strange that writing to a slot does not need any physical access.

Is this by design? in that case, why?

If my computer is attacked, someone can overwrite the key and make me lose sensitive data that may or may not have a backup. Moreover I cannot easily detect if that happened, because fake data could be used to overwrite it.

Thank you!

onlykey, 2021-04-20:
The short answer is:

What you are describing is a denial of service attack. If someone has access to your computer there are numerous other DoS attacks that could affect USB devices. However, DoS is the least of your worries if someone has access to your computer, the threat model just doesn't fit. Why would an attacker annoy you by wiping your OnlyKey data when they can keylog your passwords, conduct MiTM of web sessions, access your logged in websites etc. Someone DoSing your OnlyKey would be a welcome early warning that you need to quarantine and wipe your compromised system before something really bad happens. With that said you can enable write protection by enabling sysadmin mode.

closed 2021-04-20 by onlykey

ealfie, 2021-04-20:
Yes, of course that if someone had access to my usb device then everything is pretty screwed.
How can I enable write protection from onlykey-cli ?

ealfie, 2021-04-20:
My main concern is to prevent _myself_ from accidentally overwriting a slot having a totp secret.

In any case, I know I'm advised to backup all secrets offline