trustcrypto/OnlyKey-Firmware #124: HMAC does not work after FW upgrade with KeePassXC

pinkavaj, 2021-08-22:
Upgrade from v0.2-beta.8c
Upgrade to: latest firmware (v2.1.1-prodc)
KeePassXC version: 2.6.6

Steps to reproduce:
* Setup OnlyKey in HMAC mode to authenticate to KeePassXC
* Login to KeePassXC works
* Backup OnlyKey
* Upgrade FW using the app
* Restore configuration from backup
* Login to KeePassXC does not work any more
* Just downgrade the FW to older version allows KeePassXC authentication again

Other functions I use seems working (TOTP etc)

onlykey, 2021-08-23:
Yes, we had an announcement for this in the upgrade guide until recently. You will need to follow this guide. https://docs.crp.to/keepassxc-upgrade.html

pinkavaj, 2021-08-25:
Thx, this solved the issue for me.

closed 2021-08-25 by pinkavaj