trustcrypto/OnlyKey-Firmware #126: "space backslash char" in password field leads to stripped output

euidzero, 2021-09-15:
After upgrading to v2.1.1-prodc from beta8 the sequence : "space backslash char"  leads to stripped output :

"test\ktest" is correctly output
"test \ktest" is output as "testt"
"test \kabc" is output as "test"

FW OnlyKey v2.1.1-prodc
App : App v5.3.3

onlykey, 2021-09-15:
This is because the " \" combination is used to start sysadmin mode operation. We tried to make this a combination that is not common and not likely to interfere with any known use cases. Without sysadmin mode enabled the only options that can be used are TAB and RETURN which are " \t" and " \r".

euidzero, 2021-09-15:
It would be great to  allow the same behaviour as beta8 when sysadmin mode
is disabled. The key is advertised as acting like a  keyboard and send all
possible single  key codes. This is no more true.

Upgrading the fw broke silently existing entries containing this sequence.

Any chance to fix this in future upgrade?
Thx

onlykey, 2021-09-15:
Is there a specific requirement to be able to send these specific keys? I am not aware of a use case.

euidzero, 2021-09-15:
Coud we discuss this out of band please ?

onlykey, 2021-09-21:
We will leave this issue open to see if anyone else is affected but currently the only limitation is you can't use the `" \" `combination as a password in a slot. This is a required limitation so you can in a password specify custom keystrokes like TAB and RETURN.

closed 2021-11-26 by onlykey