trustcrypto/OnlyKey-Firmware #132: Strange behavior with FW 2.1.2

LRitzdorf, 2021-12-20:
After upgrading to firmware v2.1.2, the OnlyKey behaves unusually, both while locked and while unlocked. Unfortunately, this erratic behavior does not always occur, and is thus difficult to troubleshoot or demonstrate reliably. However, its appearance seems to be consistent within boots -- it will behave normally one boot, but may exhibit this issue on a subsequent one. If so, it will continue to do so until it is unplugged, and may or may not return to normal on the next boot.

The issue itself is as follows:

At random times while entering the unlock PIN, the LED may flash the rainbow pattern that normally occurs on boot. Any keys pressed during the flash duration are not registered, but the remainder of the PIN may be typed afterward. This is more of a nuisance than anything.

Once the key is unlocked, things get more interesting. The key operates mostly normally, except when buttons are held. For example, while attempting to enter config mode (by holding button 6 for five seconds), the LED will change from green to orange at a random point. When the button is released, the LED again flashes its rainbow pattern, and the intended long-press action does not occur (i.e. the key does not enter config mode). If you're sufficiently unlucky, this can happen while attempting to use "b" slots (1+ second presses) -- it is not limited to five-second presses. **Critically, this can prevent the user from entering config mode.**

This issue occurs on both Windows and Linux hosts (I can't test this on MacOS, unfortunately). I'm not versed in the OnlyKey implementation, but it seems as if some internal (to the OnlyKey) event triggers this, as it occurs whether or not the key is locked, and regardless of which profile (primary or secondary) the key is unlocked to. This internal event seems to occur approximately every six seconds, meaning the user has a very small window in which to execute a five-second press. However, this can be "timed" by waiting six seconds after the last interrupted press.

onlykey, 2021-12-21:
@LRitzdorf 

Thanks for the detailed explanation here. The v2.1.2 firmware has an auto-configure feature for the touch sense so in the event that there are environmental changes that change the baseline touch readings the key will reconfigure (this is the rainbow pattern). During reconfigure if there is a touch detected it has to start over with so that is why it does not register button presses. Now it should only do this on rare occasions but it seems you are seeing this quite regularly. Did you have any touch sense issues before this with FW v2.1.1?

LRitzdorf, 2021-12-21:
Thanks for the prompt response!

Prior to v2.1.2, I've had no issues with touch sensitivity – also, I've downgraded my key to v2.1.1 for the time being, and it's working normally there.

onlykey, 2021-12-21:
@LRitzdorf Unfortunately, we have not been able to reproduce the issue in our testing. As hardware devices may interact differently this is a hard one to reproduce. I have created a new firmware file and I was wondering if you would be willing to test this to see if it resolves the issue? The only change in this firmware is that it will not trigger touch sense recalibration when it detects a low condition like it does in v2.1.2 I expect this is what is causing the issue for you. Please let me know if you can try this out and if it resolves the issue you are seeing.
[Signed_OnlyKey_2_1_3_STD.txt](https://github.com/trustcrypto/OnlyKey-Firmware/files/7758009/Signed_OnlyKey_2_1_3_STD.txt)

LRitzdorf, 2021-12-22:
@onlykey I can confirm that the experimental v2.1.3 firmware works as expected. No touch recalibration occurs, of course, and long presses work as expected.

LRitzdorf, 2021-12-22:
I do have access to a second OnlyKey that I can test with v2.1.2 as soon as I have time. I'll update here when I have more information.

onlykey, 2021-12-23:
@LRitzdorf Thanks for verifying this

closed 2022-04-21 by onlykey