trustcrypto/OnlyKey-Firmware #135: Keyboard detection on Android not working with OnlyKey DUO

Vilhjalmr26, 2022-03-17:
plugging my new OnlyKey DUO into android samsung s21 5g, it is not detected as a keyboard.  using the usb a to c adapter that is part of the DUO to plug my original OnlyKey in works fine, so it is most likely something with the DUO firmware.  Duo works fine with PC.

mr-sour, 2022-03-27:
Same issue, Looks like its missing the usb header info that makes it appear as a keyboard on android.

onlykey, 2022-03-28:
@Vilhjalmr26 What kind of adapter do you use with OnlyKey that works? The firmware on OnlyKey and OnlyKey DUO is essentially the same.

Vilhjalmr26, 2022-03-28:
I took the duo out of its usb c adapter and plugged the original onlykey into it.  Had to hold it just right, as the connection was loose.

onlykey, 2022-03-28:
> I took the duo out of its usb c adapter and plugged the original onlykey into it. Had to hold it just right, as the connection between onlykey and adapter was loose.

Was the keyboard detected when you did this or not?

onlykey, 2022-03-30:
@Vilhjalmr26 What firmware version do you have on your original onlykey? Still working to reproduce this issue, I don't have an S21 to test, no luck on Samsung A32 on Android 12 everything works fine on this model.

Vilhjalmr26, 2022-03-30:
> @Vilhjalmr26 What firmware version do you have on your original onlykey? Still working to reproduce this issue, I don't have an S21 to test, no luck on Samsung A32 on Android 12 everything works fine on this model.

Original OnlyKey is on v2.1.0-prodc
OnlyKey Duo is on v3.0.0-prodn

no-sauce, 2022-04-05:
I too am having this same situation with a Samsung S20+.   The phone doesn't detect it as a keyboard.

onlykey, 2022-04-16:
@Vilhjalmr26 @mr-sour @no-sauce Please try the latest firmware here to address issue with Samsung devices - https://docs.crp.to/duousersguide.html#download-firmware

Vilhjalmr26, 2022-04-16:
> @Vilhjalmr26 @mr-sour @no-sauce Please try the latest firmware here to address issue with Samsung devices - https://docs.crp.to/duousersguide.html#download-firmware

Just loaded the firmware and tested.  Seems to work perfectly now.  Thanks so much.

no-sauce, 2022-04-17:
The new firmware has worked on the S20+ aswell thanks.