trustcrypto/OnlyKey-Firmware #136: Firmware upgrade from v2.1.0-prodc to v2.1.2-prodc with onlykey-app 5.3.5 wiped away HMAC and ECC slots

nkichukov, 2022-03-24:
Hi folks,
just upgraded my onlykey to the latest firmware and I realized that HMAC challenge response no longer works (for keepass database that uses it) and also onlykey-agent for ssh does not work as the self.skeyslot is 'NoneType' instead of 'int'. It appears that the HMAC and ECC slots have been wiped away or are otherwise inaccessible. Has anyone else experienced this?

I am thinking about restoring from latest backup, but just wanted to let you know and check in case this is a known issue or if there is a workaround other than restore from backup or adding them manually (those that were not onlykey generated internally, as those I do not have on file elsewhere).

Thank you,
-N

onlykey, 2022-03-24:
@nkichukov I am not aware of an issue with HMAC challenge response not working after upgrade from v2.1.0 to v2.1.2, however there was a change required for upgrade from beta firmware which is described here https://docs.crp.to/keepassxc-upgrade.html

For onlykey-agent the naming changed so that slot names in the agent are consistent with the app - https://docs.crp.to/onlykey-agent.html#ssh-agent-quickstart-guide-stored-keys
So instead of this command:
```
$ onlykey-agent identity@myhost -sk 102
```
You would use ECC2 instead of 102:
```
$ onlykey-agent identity@myhost -sk ECC2
```

nkichukov, 2022-03-24:
Hello Tim, thanks for the quick response. Indeed, replacing 102 with ECC2 fixed the issue for SSH.

Can we figure out what happened with the HMAC challenge response... The failure from keepassxc is:

```
Error while reading the database: Invalid credentials were provided, please try again.
If this reoccurs, then your database file may be corrupt. (HMAC mismatch)
```

onlykey, 2022-03-24:
@nkichukov I have tried to replicate this issue but no luck, you can downgrade your firmware back to v2.1.0 - https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v2.1.0-prod

nkichukov, 2022-03-24:
Hi Tim,
reverting to the old firmware 2.1.0 fixed the problem and the keepassxc database loads alright.

Let me know how to troubleshoot this further. 
Thank you.

Joeviocoe, 2023-04-16:
I'm having the same issue with the same error message.

Downgrading to firmware 2.1.0 did not work for me.

I did some testing with fresh KeePassXC databases and writing to new yubikeys.
I'm thinking that since the key that I'm using for existing databases and yubileys were originally set for yubikey hmac "Fixed 64 byte input", it may be incompatible with any version of OnlyKey.  It seems to work if I use "variable input".

@onlykey can you confirm this suspicion?

onlykey, 2023-04-18:
Are you able to install the latest firmware https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v3.0.4-prod
This should support fixed or variable input.

itoffshore, 2023-07-22:
I experienced a similar issue after upgrading the firmware from `2.1.0` => `3.0.4` & restoring from backup (after accidentally causing a factory reset). My issue seemed like a **race condition** stopping the `HMAC` button tap from working (the **KeepassXC** message to tap a button appeared & immediately disappeared)

I solved this by downgrading the firmware back to `2.1.0` / restoring from backup & touching a button before the `HMAC` race condition / issue occurred (trial & error many attempts)

For **KeepassXC** users upgrading firmware to a new **major version** it is probably prudent to:

---

* before upgrading temporarily remove the `HMAC` challenge-response security from **KeepassXC** & backup the database without `HMAC` to somewhere encrypted.
* after upgrading the firmware & restoring from backup **if you have issues** with `HMAC` / **KeypassXC** - create a random `HMAC` manually:
  - `openssl rand -hex 20`
  -  load this value into slots `HMAC1` / `HMAC2` from the '_Advanced_' tab in the OnlyKey app.
 * put the `HMAC` value in your encrypted backup with the db without `HMAC` protection. Follow the [3-2-1 of backups](https://www.redhat.com/sysadmin/5-backup-tips).
 * add the `HMAC` security back to a **copy** of the lesser protected db to become your new daily protected db.

---

This cautious procedure should stop people losing access to their data & allow slots to be regenerated if required due to an unforeseen bug. 

In the future the **KeepassXC** integration could optionally offer to show the generated `HMAC` value one time after generation so it can be backed up.

  - Alternatively the above procedure could be recommended in the docs for high safety in **major version** upgrades. 
  - The OnlyKey docs could also show guidance for **version upgrade paths** that have been well tested (similar to [Gitlab's upgrade paths](https://docs.gitlab.com/ee/update/#upgrade-paths)).

After today's restore + new `HMAC` slots - I no longer see the **KeepassXC** message to tap a button for the `challenge-response` (despite setting it as required in `config mode`) - & I wondered if in OnlyKey firmware `3.0.4`  **KeepassXC** no longer prompts for this if you use another slot for the `master` passphrase ? (I suppose most probably I should wipe & recreate the 2 x `HMAC` slots)