trustcrypto/OnlyKey-Firmware #138: OnlyKey DUO has Solid White Light

cmoore16, 2022-04-12:
Steps that led to this:
1. Forgot to put OnlyKey into config mode
2. Uploaded the new firmware and clicked "Load Firmware to OnlyKey"
3. Realizing it's not in config mode, I put it in config mode at the same time it is uploading the firmware
4. The light now switches between red and white while constantly disconnecting and reconnecting with the following messages: RECEIVED OKFWUPDATE & Error older firmware not supported
5. When I unplug and plug back in, the light remains solid white (so long as the app is closed).

First, Is there another way to perform a factory reset other than the 2 mentioned in the documentation?  Second, considering I may have completely flashed the device, is it possible to upload the firmware to the device without the app?

cmoore16, 2022-04-12:
Update:

I attempted to update the firmware on a second key and received the same result and error codes, even when following the instructions listed on the firmware tab.

cmoore16, 2022-04-12:
I just found that release and didn't realize that DUO was already on it.  Could the signed firmware txt files be provided for that now or is that only when it is fully released?  Is there a release date planned for 3.0.0?

hockey6611, 2022-04-15:
I just received my OnlyKey DUO from the Indiegogo campaign. I noted that the backup key mode and full-wipe, are enabled by default. So I think the only way to reset, is to wipe, and reload the firmware. I restored factory default, but now am unable to upload firmware with the 3.0.0 release.

hockey6611, 2022-04-15:
I commented too soon. Found the file linked in the Duo [docs](https://docs.crp.to/duousersguide.html#loading-onlykey-firmware). The key I received had already been "set up". When I loaded the onlykey program for the first time, I was not prompted by the guided setup. After reloading the firmware, it showed as new and all seems ok so far.

closed 2022-04-17 by cmoore16