trustcrypto/OnlyKey-Firmware #141: Problem with OATH-TOTP on firmware 2.1.2

bwoznicki, 2022-06-03:
There seems to be a problem with generated code for OATH-TOTP on firmware 2.1.2. I have originally setup authentication on 2.1.1 and everything was working fine ( Amazon / Github etc). Recently I have noticed that generated code never works, I have disconnected - reconnected key, wiped the slot and reconfigured again several times with no luck. Downgrading to 2.1.1 seems to fix the problem.

bwoznicki, 2022-06-03:
Yea I thought it was strange, I thought it might be the time issue, but simply downgrading fixed the issue instantly with the same setup on the slot. Might upgrade back to 2.1.2 and see if the issue persist.

matbgn, 2022-08-18:
I can only double @bwoznicki OTP generated are just totally wrong from last upgrade

bwoznicki, 2022-08-18:
Same, what worked for me originally is downgrade followed by upgrade, for some slots/sites the OTP is fine while for others just spits out wrong code. Looks like something causes it to go out of sync after a while. I have just logged off Github, logged back in and again the OTP is wrong. Happy to help with testing

onlykey, 2022-08-18:
@bwoznicki I don't see any changes to TOTP in v2.1.1 vs v2.1.2 firmware. Can you provide a TOTP secret that generates a code different than this site on your Onlykey - https://totp.danhersam.com/
*A secret that you aren't currently using for an active account

If I can replicate the issue I can fix it but so far have not been able to see any issues with TOTP. If time is correctly set on the computer where the OnlyKey app is running the app sends the time to OnlyKey and that is used to generate TOTP on device.

bwoznicki, 2022-08-19:
I dont think this is easy to replicate as it takes time to go wrong. If I reset two-factor on Github now, it will work ok for few months. I never used to save the secret so could not compare the failing one coming from Only-key to what I get from https://totp.danhersam.com/ is it possible that secret it self gets corrupted somehow on the key ? Is it possible to retrieve the secret stored on the key ?
As for the time set on the device this cant be the issue as all the generated codes would be wrong but like I mentioned before some slots are fine while others generate wrong code.

bwoznicki, 2022-10-30:
I believe i found a solution, there must be time sync problem. Just had two different OATH-TOTP failing. Closing/reopening the onlykey app / reconnecting key fixed both. FYI @matbgn

matbgn, 2022-10-30:
Yeah but if you want to rely on it for work it's not an option unfortunately.

onlykey, 2022-10-31:
@bwoznicki Glad that worked for you. If removing device and reinserting corrects issue then I suspect the issue is time drift. The OnlyKey gets the correct time from the app when you first connect device but if you were to leave the OnlyKey plugged in for weeks or months and it could have some time drift over a long period of time like this. As TOTP requires time to be within a 30 second window if device has time that is even slightly off it would require resync by removing/reinserting device.

matbgn, 2022-10-31:
It's clearly a time drift but for me it happens within a very few hours (<4)

alexhk90, 2024-12-22:
Acknowledging that this thread is a couple of years old so may no longer be an issue, but as it's still Open in case it is still an issue I just submitted a feature request that could help at least work around this time drift: https://onlykey.discourse.group/t/fido2-signcount-view-update-in-onlykey-app-cli-and-or-always-process-settime-oksettime-command-in-onlykey/1400 

If the OnlyKey CLI `settime` command was processed every time, this could be automated to be sent periodically (say, every 30 minutes) to ensure the time on the OnlyKey remained in sync (enough not to cause an issue with TOTP).

matbgn, 2024-12-22:
Still an issue in m'y perspective, so thank you 🙏