trustcrypto/OnlyKey-Firmware #143: OnlyKey buggs-out keepass database in KeePassXC when the key times-out during a challenge-response

SkyKwas, 2022-06-08:
Hi, I had a recent issue with my OnlyKey and KeePass database. In short, when the OnlyKey times-out due to inactivity during a challenge-response, KeePassXC (KPXC) experiences strange effects. I already submitted an issue in their GitHub, but they responded saying that this is an OnlyKey issue and not KPXC. They then said that I should report this here.

The issue on their side is titled:
" Database buggs-out when OnlyKey times-out during challenge-response #8129 "

Here is what I posted:
> ## Overview
> 
> When unlocking or saving a database with a hardware key as part of authentication, a successful challenge-response is needed. But, there seems to be a bug with OnlyKey. I'm using an OnlyKey and have had no issues with it working with KPXC until recently.
> 
> When KPXC requests a challenge-response, there is roughly a 10 second window where you need to tap your hardware key to pass the challenge-response. But, if OnlyKey times-out during this small window, the database experiences effects.
> 
> For those that don't know, you have to type in a PIN after plugging an OnlyKey in a computer to use it. After the PIN is entered correctly, the LED on it will glow solid green, meaning that it is ready to be used. When using it on KPXC for a challenge-response, the OnlyKey will flash yellow, meaning it is waiting for someone to physically touch it. When the key is touched during this yellow flashing period, the challenge-response has been completed and KPXC opens the database.
> 
> If the key is not touched, the challenge-response fails and you need to try to unlock the database again through KPXC.
> 
> If the OnlyKey remains inactive for 30 minutes (this is the default), it locks itself and the LED turns off. The PIN will need to be re-entered to use it again.
> 
> Now that I've explained all of that, here is the issue: One time when I was trying to save changes to a database, I needed to pass the challenge-response. KPXC was waiting for me to tap my key and the OnlyKey was flashing yellow. As I was reaching for the key, the flashing yellow light stopped and I tapped it.
> 
> The OnlyKey just so happened to time-out after 30 minutes of inactivity within that ~10 second challenge-response window.
> 
> KPXC said the challenge-response failed. After looking at the key, the LED is still off. I pulled the key out and put it back in and entered my PIN to get the solid green light.
> 
> When I tried the challenge-response again, KPXC said it failed. I tried a few more times before restarting my computer to see if that will fix it.
> 
> When I booted back into my computer and tried to open the database, KPXC does something even stranger: the database opens... without even taping the key. And it stayed this way...
> 
> Now, the database still needs the key to be active (solid green LED), but KPXC completely bypasses the "touch your key" requirement. All I did was typed in my password, selected the key, select "unlock", and... it instantly unlocked the database. In addition, any changes I made to the database instantly saved without any interaction with the key.
> 
> At this point I thought I corrupted my database and needed to make a new one, but I tried one more thing. Sometimes, the OnlyKey does do the "yellow flashing" like it's suppose to. When this happened, I purposefully let the ~10 second window time-out. And... strangely enough, that actually seemed to revert all apparent problems I was having.
> 
> I have been using the same database for ~4 weeks now and have not had any strange things happen.
> ## Steps to Reproduce
> 
>     1. Make a database with password + challenge-response with an OnlyKey
> 
>     2. Make OnlyKey time-out within KPXC's challenge-response window (the default 30 minutes can be changed in their app)
> 
> 
> KeePassXC - 2.7.0 (The issue may still be present in the current version) OnlyKey - 2.1.2
> 
> Operating System: Linux Mint
> Desktop Env: Cinnamon
> Windowing System: X11

onlykey, 2022-06-10:
@SkyKwas So when the 30 minute inactivity occurs and OnlyKey locks itself its not actually locking. It is doing a hardware reboot just as if you had removed/reinserted OnlyKey. Can you recreate your issue if you physically remove OnlyKey during challenge-response? If so then the issue would likely be something in KeepassXC where there needs to be added support for removing security key during challenge-response request.

onlykey, 2022-06-10:
> When I booted back into my computer and tried to open the database, KPXC does something even stranger: the database opens... without even taping the key. And it stayed this way...

For this I suspect the app might have some kind of fail safe where if the challenge-response fails and the app crashes it saves the database with just the password rather than not saving the database (you lose the data you are trying to save).

SkyKwas, 2022-06-12:
I tried pulling out the key during the challenge-response, but I couldn't recreate the issue.