trustcrypto/OnlyKey-Firmware #148: Can't enter config mode

Derton1, 2022-07-11:
When I hold down #6 for 5+ seconds on my OnlyKey to enter config mode it doesn't work. It blinks twice red, then goes back to the standard flashing LED, the light doesn't turn off or anything. I tried updating the firmware but since you also need to be in config mode to do that it did not work. I tried entering config mode on Linux, Windows, and MacOS, none of them worked. Just the red LED flashing twice then nothing.

lewishazell, 2022-07-13:
Not a maintainer here but I wanted to ask if you have a PIN set? The device needs you to enter the PIN once again (after the initial unlock and holding down the button) before it will enter config mode.

I thought I'd ask because I had this problem when I first tried to enter config mode!

Derton1, 2022-07-14:
Thanks for the reply, I do have a PIN set yes. After I hold down the config button it doesn't prompt me to enter the pass code again. The key flashes red twice and falls back into operating like a password manager, no prompt to enter the PIN.

bryanpedini, 2022-07-16:
Just to make sure you're using the correct procedure...
To put your OnlyKey in config mode you need to:
- Connect the OnlyKey to a USB port to your PC (dah)
- Input your **primary** PIN code
- Wait until you see a steady green light (not flashing)
- Hold button `#6` for 5 to 6 seconds (a little more, like half a sec, is never a bad idea, just to make sure)
- Re-input **again** your **primary** PIN code
- Wait for a slow flashing red light

If you're following the abovementioned correct procedure and you're still incapable of entering config mode, try to reset your OnlyKey to factory defaults (after maybe having performed a backup), or you might have a corrupted firmware (somehow).