trustcrypto/OnlyKey-Firmware #151: Failed to restore FIDO2 credentials from backup

pinkavaj, 2022-07-31:
I have 2 OnlyKey tokens, both share the same key for backup/restore. I have tried to transfer the FIDO2 credentials using backup/restore, but the ssh-key stored in the credentials is in non-usable state after the restore.

Steps to reproduce:
* Unlock the OnlyKey 1
* Set pin using `onlykey-cli set-pin`
* create new credential using `ssh-keygen -b 521 -t ecdsa-sk -O resident -f ~/.ssh/id_ecdsa_resident` (press 1 to allow upload)
* Verify the key is avaiable and can be used by ssh eg `onlykey-cli credential ls` shows 1 item, `ssh-add -K` works as expected
* Create backup of the key context using `#1` button and store to file backup.txt
* Remove OnlyKey 1, Unlock OnlyKey 2
* Set to config mode and restore from `backup.txt`
* Unlock again after restore
* Try to use the ssh-key ends up with followig error:

```
ssh-add -K
Enter PIN for authenticator:
Provider "internal" returned failure -1
Unable to load resident keys: invalid format

onlykey-cli credential info
PIN:
Existing resident keys: 1
Remaining resident keys: 11

onlykey-cli credential ls
PIN:
# Empty result, no key listed
```

onlykey, 2022-08-01:
@pinkavaj We have received reports of issues with backup and restore of FIDO2 resident keys. We are looking to address this in the next firmware release.

niko-lay, 2022-08-06:
I've faced same issue with 2 onlykeys and WebAuthn as second factor, same scenario backup 1st and restore to 2ns one.
On some sites both onlykeys works fine (https://privateemail.com/), on other  (https://github.com/ and aws sso) only one could be used.

onlykey, 2022-08-06:
@niko-lay This sounds like a different issue. With OnlyKey security keys you can have a primary and a backup key. Both keys are not meant to be used at the same time because one key will inevitably be out of sync, or the counter of that key will be lower than the expected counter for the next authentication. In order to correct the counter of the backup key to make primary it needs to connect to the OnlyKey App.

pinkavaj, 2024-08-27:
Works for me now, no issue any more, closing.