trustcrypto/OnlyKey-Firmware #152: Update udev rules to include GROUP

TheOnAndOnlyZenomat, 2022-09-14:
I don't know if this is the right place, but the PKGBUILD from the `onlykey-udev` AUR package, pointed to a rawgithubuserconent link under the trustcrypto user, under which I couldn't find a repo containing those rules, so I am opening this issue here.

With some commands, like `onlykey-gpg init` or `onlykey-agent -c` you get an error like this:
```
2022-09-14 14:15:37,533 ERROR        failed to connect                                                                                    [client.py:279]
Traceback (most recent call last):
  File "/home/zeno/.local/lib/python3.10/site-packages/onlykey/client.py", line 270, in _connect
    self._hid.open_path(self.path)
  File "hid.pyx", line 142, in hid.device.open_path
OSError: open failed
2022-09-14 14:15:39,135 ERROR        failed to connect                        
```
In my case this was caused, because the udev rules didn't specify a group. After I updated them to this:
```
SUBSYSTEMS=="usb", ATTRS{ID_VENDOR_ID}=="1d50", ATTRS{ID_MODEL_ID}=="60fc", GROUP="plugdev", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{ID_VENDOR_ID}=="1d50", ATTRS{ID_MODEL_ID}=="60fc", GROUP="plugdev", MODE:="0666"
```
it worked.
Note, I also updated `idVendor` to `ID_VENDOR_ID` as this is listed using the `udevadm monitor` command, and till now, that didn't cause issues.

bryanpedini, 2022-10-17:
> a rawgithubuserconent link under the trustcrypto user, under which I couldn't find a repo containing those rules

probably a [gist](https://gist.github.com) then

TheOnAndOnlyZenomat, 2022-10-17:
I know I could put them in a gist, but because the linked gist is the main source for the udev file, I want the updated information in that gist, and not one I open myself

onlykey, 2022-10-17:
We have a link to the UDEV rule here - https://docs.onlykey.io/linux.html

I think that is the same link used by the AUR. The AUR was not made by us but by an OnlyKey user contributor.

TheOnAndOnlyZenomat, 2022-10-21:
It is the link used in the AUR, but I thought, that you were in control of gist. And even if not, it would make sense to update the link in the docs to a new gist, or even a file in a repo, which can be easier updated, than a gist, containing the changes.
But maybe I am just missing something here, so sorry, if I am talking stupid stuff :D