trustcrypto/OnlyKey-Firmware #153: Passive USB fingerprint self-destruct

HarryR, 2022-10-07:
Rather than having a 'self-destruct' pin, I want the OnlyKey device to self-destruct if it's plugged into any device other than the one it has been bound to.

This can be achieved through some kind of fingerprinting, additionally per-device HTOTP could be derived using this fingerprint meaning the same drive (e.g. USB-C attached NVEMe and same OnlyKey dongle) on another computer will not only fail to unlock but will trigger a kill-switch.

In combination with TPM, OPAL and UEFI SecureBoot this will address many edge cases where OnlyKey is currently vulnerable to physical attacks.

onlykey, 2022-10-07:
Some considerations for this feature request:

- How do we prevent inadvertent self-destruct? I.e. A child plugs in your device and wipes it.
- What kind of fingerprinting would be used that couldn't be spoofed?
- OnlyKey is not designed to have a battery/capacitor, this is not something we plan to add

We typically implement features that have the widest range of use and interest from a large number of users. I will leave this open, feel free to add to this if there is interest.

HarryR, 2022-10-07:
> How do we prevent inadvertent self-destruct? I.e. A child plugs in your device and wipes it.

Given that the device is now 'tamper-evident', that would be a discussion between the hypothetical parent and child.

> What kind of fingerprinting would be used that couldn't be spoofed?

Other devices on the bus, the MACs etc. - it's not important that it can't be spoofed via some highly pre-orchestrated plan, only that if there's a significant probability that the device it's plugged into doesn't match the device which it was bound with - then it should factory reset.

> OnlyKey is not designed to have a battery/capacitor, this is not something we plan to add

Understood.

HarryR, 2022-10-07:
However, having an 'ohshit' pin that I could solder to something interesting would be... especially if it can be activated via firmware