trustcrypto/OnlyKey-Firmware #157: Keymap for ^ on GERMAN_MAC

Piggeldi2013, 2022-10-28:
The keymap for the ^ sign on Keyboard Layout GEMAN_MAC doesn't seem to work correctly.

If used in a password, the ^ sign is replaced with a space.

This seems to be the only thing that isn't working in the keymap.

So, given a password of: 
asd^dde
results in:
asd dde

Setting of the password has been tested with the App and also on command line - with same results.

Please advise.

onlykey, 2022-10-28:
@Piggeldi2013 Can you use the app here and let us know what keycode you are getting when you type ^ with your keyboard and then what keycode you get when OnlyKey types this?

https://apps.apple.com/us/app/key-codes/id414568915?mt=12

Piggeldi2013, 2022-10-28:
[image]

Same for both - adding it in the App, and also with cli.

onlykey, 2022-10-28:
No I mean if you type the ^ character via your keyboard like this:
[image]

Then 2nd screenshot of if OnlyKey types ^ (set your password as this in a slot for the test)

Piggeldi2013, 2022-10-28:
Thanks for coming back.

The key (pressed at keyboard) is:
[image]

And the image in my post before shows how the "password" is sent by the OnlyKey.

onlykey, 2022-10-28:
Can you send the whole thing like, type "^ ^" via keyboard and OnlyKey like this:

^ ^ typed on keyboard:
[image]

^ ^ typed via OnlyKey
[image]

Piggeldi2013, 2022-10-29:
Sure I can - here the screens for ^ ^

Keyboard: 
[image]

OnlyKey:
[image]

onlykey, 2022-10-31:
@Piggeldi2013 Oh sorry it looks like you don't have the "Log key up events and modifier changes" box checked, can you send the same results again with that box checked?

Piggeldi2013, 2022-10-31:
Hi, no problem - I'm glad that I get help and can give some more infos - thanks for that!!

Keyboard:
[image]

OnlyKey:
[image]

Piggeldi2013, 2022-11-09:
Any news on that? Something I can help or provide?

onlykey, 2022-11-29:
@Piggeldi2013 No, thank you for the help. It looks like we can generate the ^ character using the combination of SHIFT+ALT+6

We will get this change into the next release

onlykey, 2022-12-06:
@Piggeldi2013 Can you please test the firmware here? I believe the issue is fixed now.
[Signed_OnlyKey_3_0_4_STD.txt](https://github.com/trustcrypto/OnlyKey-Firmware/files/10170347/Signed_OnlyKey_3_0_4_STD.txt)