trustcrypto/OnlyKey-Firmware #160: Enhancement - function as an OpenPGP smartcard according to 3.4 standards.

gibsonjareds, 2023-02-21:
Hello!

I’m opening this issue because I noticed an issue when trying to integrate the Onlykey into my existing GPG workflow. 

The existing `onlykey-agent` is great, but it creates some headache when moving between multiple machines. I’ll try to explain the two main scenarios to the best of my ability:

## 1 - Bootstrapping. 

Onlykey-agent and the onlykey-cli require a significant amount of dependencies to be available that may be difficult to work with depending on how you’re using your PGP keys.  

A common workflow, for example, is generating a new key pair on an air gapped computer, preferably on a liveUSB or other isolated system. These images usually have GPG or some other implementation installed, but can lack Python support, and obviously don’t come with the onlykey tools. 

If the only key was able to function as a smart card, this would eliminate this problem as it would work with most OpenPGP implementations out of the box. It would also allow me to perform pgp-reliant tasks on machines that I don’t necessarily have time to bootstrap. 

## 2 - Key management. 

When using a physical hardware key, it’s pretty typical to want to generate auth or signing keys on individual devices that can be used without the hardware key. This is great for “revoking” keys that are compromised or that exist on machines that are no longer used.

The onlykey-gpg implementation seems to make it impossible to generate non-hardware subkeys, and the fact that it requires me to have a separate GNUPG home makes it more difficult to manage my existing trust DB. 

Would the team be open to PRs for this functionality? Is the smart card standard otherwise incompatible with the device/firmware as they are currently implemented? I don’t want to rehash something that you’ve already internally discussed, but rather wanted to start a conversation around this topic to see if it could go anywhere.

gibsonjareds, 2023-02-21:
Following up, I just realized that you had closed #40 (which I somehow missed in my search) with no resolution. Is the sentiment expressed there still the sentiment of the OnlyKey team?

closed 2023-02-21 by gibsonjareds

onlykey, 2023-02-22:
@gibsonjareds Yes, we currently are only pursuing OpenPGP support through the OnlyKey Agent app here - https://docs.onlykey.io/onlykey-agent.html

Implementation of smartcard functionality would not be feasible at this time or in the near future.