trustcrypto/OnlyKey-Firmware #169: Backup

haplo, 2024-01-22:
I hold button 1 for 5 seconds to write a backup, the header gets printed:

```
-----BEGIN ONLYKEY BACKUP-----
```

But it hangs after that and never finishes.

I have tried in two different systems, one with Wayland another with Xorg. I also tried on a text console.

haplo, 2024-01-26:
An update: I was lucky enough to have 2 other OnlyKeys that I could restore previous backups to so I decided to upgrade the firmware of the faulty one.

After upgrading to firmware 3.0.4 the OnlyKey will still not write a backup, same issue.

I will try to reset it to factory and restore a backup, see if that fixes it, but first I need to check that I will not get locked out of anything as this was my main OnlyKey.

notlaast, 2024-08-16:
Hi, any results after resetting to factory and restoring from backup? I just received an OnlyKey (international travel edition), and flashed to standard firmware (v3.0.4), added some keys, and finally now I realize I have this same issue about displaying only the header.

For what it's worth, the light switches to a constant green+red. From what I can tell, the only way to return the OnlyKey back to a functional state from here is to unplug it.

Tipoff4317, 2024-08-18:
@notlaast Not the OP, but there is this thread, if it's related to you:

https://onlykey.discourse.group/t/onlykey-hangs-on-backup/1310/4

notlaast, 2024-08-19:
I can confirm that the steps outlined under https://onlykey.discourse.group/t/onlykey-hangs-on-backup/1310/7 worked. That is, I self-destructed the key and set it back up, and everything now appears functional.

Maybe I had made a mistake during initial setup and something was in a poorly defined state, who knows.

Perhaps you could close this ticket @haplo ?