trustcrypto/OnlyKey-Firmware #170: Will there be any more updates to the firmware?

MNylif, 2024-02-04:


EnlightenedBacon, 2024-02-27:
This is concerning. I was just at the website to buy one and thought I'd get out the git repo. If the firmware is from 2022 and there's no activity, does the company still even operate?

onlykey, 2024-02-27:
There will be future firmware releases as needed. Currently this is the latest firmware https://github.com/trustcrypto/OnlyKey-Firmware/releases/tag/v3.0.4-prod

YellowOnion, 2024-03-22:
@onlykey your crp.to domain is returning 503's, PITA for new users, why not just have one?

FormBurden, 2024-05-15:
Even if for two years there is no need for a firmware update, I would suggest either updating on your website (main website, the news is also from 2022, not good) or on the Description of this Repo that the project has not been abandoned and the company is still in operation. Having no updates on your site and the firmware from 2022 does not boast confidence in your product. I came here to see if their was an update to my OnlyKey, only to find out that I have the latest still. Please give some updates of anything.

closed 2024-05-15 by onlykey